raspberrypi/documentation #937: Is the USB connectors' shield connected to GND?

jacobq, 2018-07-03:
I realize that full schematics are not available, but could someone at least tell me where to find documentation about how the USB ports are grounded? I'm seeing some unexpected behavior when I connect certain devices and suspect it may be related how the shield is or isn't connected.

([Here's a related issue](https://github.com/raspberrypi/documentation/issues/437), but it doesn't answer this question.)

JamesH65, 2018-07-03:
@Roger-Thornton Ping.

jacobq, 2018-07-03:
Aww, now I feel bad for bugging him on Twitter ;) Thanks in advance.

Roger-Thornton, 2018-07-03:
Can you tell me what product this relates to and which USB port please. 

-Roger

jacobq, 2018-07-03:
@Roger-Thornton We're using an RPi3 Model B+ (latest) and connecting a 3rd party device to any one of the 4 USB host ports (have not been specifying a particular one to use but we can if it matters). We think the device may have incorrectly grounded the shield, forming a loop but cannot confirm without more info regarding the wiring of the RPi's USB shield (though I suppose I could get some probes out...).

Roger-Thornton, 2018-07-05:
All the shields on the USB connectors are connected to Ground, the same as
on all Pi boards.

closed 2018-07-05 by jacobq