raspberrypi/documentation #938: Why official documentation forward users make to another web sites??

zervankoru, 2018-07-04:
For example, https://www.raspberrypi.org/documentation/configuration/raspi-config.md - at the end: This article uses content from the eLinux wiki page RPi raspi-config, which is shared under the Creative Commons Attribution-ShareAlike 3.0 Unported license

Same thing here: https://www.raspberrypi.org/documentation/configuration/config-txt/README.md

XECDesign, 2018-07-04:
1) Attribution is part of the license.
2) It's nice to acknowledge the work of others if you're building off it.

lurch, 2018-07-04:
Because (parts of) the official documentation were based (in part) on the community-written documentation on the eLinux wiki, and the footer you've quoted is just acknowledging that fact & respecting the Creative Commons license. There's no intention to "forward users to another website".

zervankoru, 2018-07-04:
@XECDesign but how have they done it? Isn't official documentation primary source?

XECDesign, 2018-07-04:
elinux was the first place people started gathering information about the pi. The official documentation came along much later.

zervankoru, 2018-07-04:
@XECDesign so Raspberry Pi have developed Pi, but elinux had understood, how this device works faster?

JamesH65, 2018-07-04:
@zervankoru No, they just had more time and people in the early days to collate information. When started RPF was a very small charity. Now it's much larger, and we now have enough staff to provide much better documentation.

JamesH65, 2018-07-04:
Worth adding that in many cases the Elinux stuff is now out of date, effectively superseded by the Official docs.

Since this isn't a issue report on the current docs, I'm closing it.

closed 2018-07-04 by JamesH65

bennuttall, 2018-07-04:
Here's a statement from [April 2014](https://www.raspberrypi.org/blog/welcome-to-our-new-website/) when we launched the docs:

> We have made a big change to the way we deal with documentation. A bit of background is necessary here. Until now, we’ve relied on the third-party, crowd-sourced wiki at eLinux. This was set up in 2012 when we had absolutely no staff, and we asked the community to help populate it, because we didn’t have the resources ourselves. We at the Foundation have no oversight over that wiki, and we’ve noticed that it’s become a bit out of date.
>
> So we’ve taken the decision to move all of our documentation in-house, but we’ve done so in a way that means that you can make additions and alterations if you think we’ve missed something – with our oversight. All of our documentation is written in Markdown, and lives on GitHub. It’s not an open wiki, but if you want to make a change, please open an issue on GitHub. (Learn more here.) We’ll consider all issues which are opened, and if we accept yours, you can file a pull request with your change. It’s a way to keep things lean, consistent and accurate. Everything gets looked over by the team of people who make the Raspberry Pi to be checked for accuracy: at the same time, it allows you to pull us up on anything you think we should expand on.
>
> Some of what we have here now is based around a kernel of documentation from the old eLinux wiki, and we are very, very grateful to everyone who contributed materials to it that we have been able to use here. The new documentation also covers all the stuff we used to host here separately: datasheets, hardware specs and so on. We’ve still got some editorial work to do on some of what we’ve pulled in to the new documentation, but it should be usable from today.

The Pi was launched with almost no documentation. If we'd have waited, we'd have never launched the product, and the docs wouldn't have been perfect anyway (they never will be). They were crowd-sourced from the beginning, that's why they're as good as they are.

We wanted there to be a canonical source for our official documentation, and as some of it had already been written/collected over on enlinux (which has a Creative Commons licence), it was only fair to attribute the source. It's only what we would expect if someone used documentation from here elsewhere (although we wouldn't really enforce this too strongly).