raspberrypi/documentation #944: USB Boot Mode OTP Bit Programming Fail

psymonwhite, 2018-07-17:
Using the USB Boot Mode instructions at: 

https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md 

If the OTP Bit fails to be programmed - what is the corrective action to fix this? 

The documentation states that it can fail, but does include a resolution path.

JamesH65, 2018-08-08:
@ghollingworth Is there any corrective action available?

ghollingworth, 2018-08-09:
Although in 'theory' it can fail, we'd not expect to see this happen unless there is a hardware problem or you're Pi in one in ten million...

No there's no 'corrective action'

Gordon

JamesH65, 2018-08-09:
I can update the text. Currently " If it is not, then the OTP bit has not been successfully programmed." to something like "If the bit has not been programmed, go through the programming procedure again. If the bit is still not set this would indicate a HW fault with the Pi itself."  ??

JamesH65, 2018-08-31:
@ghollingworth OK to go ahread with change suggested in previous post, or shall we just leave it?

ghollingworth, 2018-08-31:
Yes that sounds fine

JamesH65, 2018-09-12:
#982 Combined with some other changes to the same page.

JamesH65, 2018-09-12:
Fixed by #982