raspberrypi/documentation #951: Peripheral address page needs more info

tlhackque, 2018-07-31:
[Peripheral Addresses](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/peripheral_addresses.md) instructs the reader to use a couple of function calls to obtain the bus address of the peripheral page, its size, and the address of SDRAM.

This is fine - except that the page doesn't specify what header file and library contain these functions.

These should be added to the page.

lurch, 2018-07-31:
Tracing back through linked issues, I eventually found #158 ...

tlhackque, 2018-07-31:
That's where the code lives on GitHub.  But anyone who wants to use it needs to know:

- What package, if any to install (apt-get foo)
- What header file to use (#include <foo.h>, any -I if it's not in a standard library)
- What object library/sharable image to link against (-l)

Think what you get for a standard library - e.g. man sqrt produces:
````
SQRT(3)                    Linux Programmer's Manual                   SQRT(3)

NAME
       sqrt, sqrtf, sqrtl - square root function

SYNOPSIS
       #include <math.h>

       double sqrt(double x);
       float sqrtf(float x);
       long double sqrtl(long double x);

       Link with -lm.
````

Note the #include and "link with"; this example doesn't require a package.

Unless the idea is that people clone that repository - which works against the idea of a library.  Even in that case, the page should say so...

I don't mean to be picky, but as is, the page doesn't tell me what I need to know...

lurch, 2018-07-31:
I'm not disagreeing, I was just attempting to provide extra info in the meantime in case you needed it :-)

JamesH65, 2018-08-07:
Good point. Will do that.

JamesH65, 2018-08-07:
Fixed by https://github.com/raspberrypi/documentation/pull/961

tlhackque, 2018-08-07:
RE: the commit.  If you use -I to compile, the #include should just be the filename.  E.g.
#include <bcm_host.h> This is the most portable/readable approach.  If you use a full path in the #include, you don't need the -I - this is not recommended.

(In any case, you're missing the "h" in ".h" of your include.)

JamesH65, 2018-08-07:
Good point, but you do need the -I otherwise some sub includes cannot be found.

I'll update to just use the filename.

JamesH65, 2018-08-08:
Merged.

closed 2018-08-08 by JamesH65