raspberrypi/documentation #960: OMXPlayer uages page

JamesH65, 2018-08-07:
Turns out there doesn't appear to be a page dedicated to OMXplayer, which seems rather a omission. Couple of pages "How to play audio" and "how to play video" but both very short and sweet.

bennuttall, 2018-08-07:
Pasted from email:

One reason I'd keep them separate is the titles are pretty much answers to questions someone might google, e.g. "how to play audio on raspberry pi" - if they don't know what omxplayer is they're not going to include it in their search terms.

An option could be to leave those pages up and say "play audio with omxplayer - see <link>".

JamesH65, 2018-08-10:
Started this here https://github.com/raspberrypi/documentation/tree/JamesH65-omxplayer/usage/omxplayer

bennuttall, 2018-08-20:
Can we put this in Raspbian as it's more documentation of a Raspbian application than usage which is supposed to be just getting started

JamesH65, 2018-08-20:
Of course, still a work in progress, can go anywhere. I do think we may need to add some more top level categories at some point.

JamesH65, 2018-08-31:
Any idea how I move the location of a file in a PR in the github UI?

bennuttall, 2018-08-31:
Interestingly I've just found that you can...

Edit the file you want to move. Edit the filename to simply rename it:

[image]

Add `/something/` at the beginning to nest it within a directory:

[image]

after you add the slash:

[image]

Add `../` at the beginning and it actually drops the current directory from the path:

[image]

after you add the slash:

[image]

Add the intended directory name to set the new location:

[image]

after you add the slash:

[image]

JamesH65, 2018-08-31:
I knew I had seen it done, but hadn't realised you needed to be editing it to move it!

JamesH65, 2018-08-31:
So it turns out there is already an OMXplayer page, which has almost exactly the same content as I produced here. I cannot figure out how I missed it....

https://www.raspberrypi.org/documentation/raspbian/applications/omxplayer.md

bennuttall, 2018-08-31:
Oh... I wonder who wrote it... https://github.com/raspberrypi/documentation/commits/master/raspbian/applications/omxplayer.md

I think you need a holiday, James

JamesH65, 2018-08-31:
After being on holiday with the family, I do need a holiday. Closing.

closed 2018-08-31 by JamesH65