raspberrypi/documentation #963: apt-get dist-upgrade failing

gooddosh, 2018-08-08:
I have a SanDisk Ultra 128GB microSDXC UHS-I Class 10 Card installed.

I run a CrushFTP server on this Pi. The disk write speed appears to average 1.2 to 1.6 MB/s, which is excruciatingly slow.

It may also be the reason why dist-upgrade is failing. I also get kicked off SSH when it stalls and fails.

Uploads to CrushFTP fail, because the disk write speed is so painfully slower than the RAM buffer that CrushFTP uses.

The write speed of the Pi makes practical use a joke, at least for me so far.

gooddosh, 2018-08-08:
Disregard earlier references to apt-get. It worked. It was just REALLY slow due to the write speed problem.

bennuttall, 2018-08-08:
Please do not report Pi troubleshooting questions here. This is for reporting issues with the documentation contents. We provide forums for getting technical help: https://www.raspberrypi.org/forums/

closed 2018-08-08 by bennuttall