raspberrypi/documentation #970: inconsistencies in the documentation

chwe17, 2018-08-24:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/frequency-management.md
first:

> MEASURING TEMPERATURE
> Due to the architecture of the SoCs used on the Raspberry Pi range, and the use of the upstream temperature monitoring code in the Raspbian distribtution, Linux-based temperature measurements can be inaccurate. There is a gencmd that can provide an accurate and instantaneous reading of the current SoC temperature, as it communicates with the GPU directly:
> 
> vcgencmd measure_temp

and second:

> For Raspberry Pi 3 Model B+, the PCB technology has been changed to provide better heat dissipation and increased thermal mass. In addition, a soft temperature limit has been introduced, with the goal of maximising the time for which a device can "sprint" before reaching the hard limit at 85'C. When the soft limit is reached, the clock speed is reduced from 1.4GHz to 1.2GHz, and the operating voltage is reduced slightly. This reduces the rate of temperature increase: we trade a short period at 1.4GHz for a longer period at 1.2GHz. By default, the soft limit is 60°C, and this can be changed via the temp_soft_limit setting in config.txt.

is in conflict with:
https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

> Monitoring core temperature
> To view the Pi's temperature, type: cat /sys/class/thermal/thermal_zone0/temp. Divide the result by 1000 to find the value in Celsius.

and second this needs at least some clarification, due to not true for the 3B+ with the newest firmware:

> With firmware from 12th September 2016 or later, when the core temperature is between 80'C and 85'C, a warning icon showing a red half-filled thermometer will be displayed, and the ARM cores will be throttled back. If the temperature exceeds 85'C, an icon showing a fully-filled thermometer will be displayed, and both the ARM cores and the GPU will be throttled back.
> further from the overclocking page:
> Clocks relationship
> The GPU core, CPU, SDRAM and GPU each have their own PLLs and can have unrelated frequencies. The h264, v3d and ISP blocks share a PLL. For more information see here.

and further:
> To view the Pi's current frequency, type: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq. Divide the result by 1000 to find the value in MHz.

is in conflict whit @ShiftPlusOne statement on this thread (https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=219358#p1347062):

JamesH65, 2018-08-31:
Fixed by #975

JamesH65, 2018-10-01:
Fix merged.

closed 2018-10-01 by JamesH65