raspberrypi/documentation #971: GPIO current control

seamusdemora, 2018-08-25:
[I have read that limits may be programmed for the output current of GPIO pins](https://www.scribd.com/doc/101830961/GPIO-Pads-Control2). However, after reading several online versions of Broadcom's data sheet for the SoC, I've found nothing to corroborate the data in this reference. Is there a "definitive source" for this **GPIO current limit function**, either in hardware documentation *OR* software documentation?

JamesH65, 2018-08-31:
That scribd document is correct. Not sure why it's not in the peripheral specification, or where that linked  document originated though.

lurch, 2018-09-11:
IIRC there were a bunch of additional mini-documents added to scribd by Gert. See also the "More From Anonymous I42Heqce7f" towards the bottom of that page. I think scribd itself has changed quite a bit since Gert was using it (e.g. you used to be able to download the PDFs without registering, which it seems you now can't do).

JamesH65, 2018-09-12:
Ah, thanks. We should probably pull all that stuff in to our docs somehow rather than relying on scribd.

JamesH65, 2019-01-18:
This doc https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md shoud be updated to link to our own version of the scribd doc, or just insert the relevent bits in to the page itself.

seamusdemora, 2019-03-31:
Sorry to be so slow following up. Thank you for responding, and making this crystal clear.

seamusdemora, 2021-06-09:
@JamesH65 :
> This doc https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md shoud be updated to link to our own version of the scribd doc, or just insert the relevent bits in to the page itself.

The [document](https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md) you referenced is valuable as a reference for the GPIO *Electrical Characteristics* (ref the "Voltage Specifications"). However, it must be noted that as it refers to the **BCM2835**, some of these specifications ***may be*** *out-of-date*. 

This document also contains a statement that these Voltage Specifications "[were] extracted from the Compute Module datasheet [here](https://www.raspberrypi.org/documentation/hardware/computemodule/datasheet.md)." Following this link proved useful in that the [CM4 document](http://datasheets.raspberrypi.org/cm4/cm4-datasheet.pdf) contains ***related*** data (ref `3.3. Electrical Specification`, pp 15-17. 

However, in reviewing the CM4 document, two questions arise: 

1. Does the CM4 document accurately represent the RPi 4B GPIO? REF the statement *"The design of the CM4 is loosely based on the Raspberry Pi 4, Model B"* 

2. The [current GPIO document](https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md) contains numerous details that are not covered in the [CM4 document](http://datasheets.raspberrypi.org/cm4/cm4-datasheet.pdf) - For example, [*Drive Strength*](https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/gpio_pads_control.md).   **Can it be assumed that details covered in the current document for the BCM2835 *still apply* to the BCM2711?**

JamesH65, 2021-06-11:
1. yes
2. yes

closed 2021-06-11 by aallan

seamusdemora, 2021-06-11:
@JamesH65 
> Ah, thanks. We should probably pull all that stuff in to our docs somehow rather than relying on scribd.

That's still a great idea. :)

aallan, 2021-06-11:
> > Ah, thanks. We should probably pull all that stuff in to our docs somehow rather than relying on scribd.
>
> That's still a great idea. :)

There are no links from that page to scribd?

seamusdemora, 2021-06-11:
@aallan 
AFAICT, everything on scribd is now behind a paywall.

JamesH65, 2021-06-12:
I think this page https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/gpio_pads_control.md is the one that used to be in scribed.