raspberrypi/documentation #972: Sense HAT "factory hardware misconfiguration" Howto fix it?

RogerTarani, 2018-08-28:
Hi,
I encounter the fllowing trouble, connecting a Pi3B with a brand new Sense HAT :
only the IMU  (0x1c et 0x6a), P/T  (0x5c) and humidity/T sensors (0x5f) are accessible as proven by a provided software:
$ RTIMULibDrive11 
(measures are continuously displayed until Ctrl-x)

But the LEDs (and Joystick) managed by Atmel ATTINY88 are not accessible (0x46 ; to be checked - has the  Joystick the same I2C address as LEDs or an individual address, that could be inferred as 0xf3, considering some articles):
$ i2cdetect -y 1
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: — — — — — — — — — — — — —
10: — — — — — — — — — — — — 1c — — —
20: — — — — — — — — — — — — — — — —
30: — — — — — — — — — — — — — — — —
40: — — — — — — — — — — — — — — — —
50: — — — — — — — — — — — — 5c — — 5f
60: — — — — — — — — — — 6a — — — — —
70: — — — — — — — —

So, obviously, the library sense_hat.py is unhappy when it tries to access LEDs (SENSE_HAT_FB_NAME), for instance in the very first example Hello Wordl!  at [https://projects.raspberrypi.org/en/projects/getting-started-with-the-sense-hat/4](url):
$ python main.py
Traceback (most recent call last):
File “main.py”, line 2, in
sense = SenseHat()
File “/home/pi/Documents/SenseHAT/Helloworld/sense_hat/sense_hat.py”, line 39, in __init__
raise OSError(‘Cannot detect %s device’ % self.SENSE_HAT_FB_NAME)
OSError: Cannot detect RPi-Sense FB device

So, it seems clear that there is a HW misconfiguration with my/the Sense HAT.

I tried another Pi3B.
I tried things like add the following line to /boot/config.txt :
dtoverlay = rpi-sense

I tried to update the EEPROM of the Sense HAT but it failed (done from a Pi3B)
[cf. https://www.raspberrypi.org/documentation/hardware/sense-hat/](url)
“These steps may not work on Raspberry Pi 2 Model B Rev 1.0 and Raspberry Pi 3 Model B boards. The firmware will take control of I2C0, causing the ID pins to be configured as inputs.”

Then I found this long discussion: 
**I followed the instructions to update EEPROM and it bricked my Sense HAT #326** 
[https://github.com/raspberrypi/documentation/issues/326](url) 
On march 15, 2016 Simen told us that he could successfully flash the EEPROM of the Sense HAT from a Pi2 and got it operational when put on his Pi3.
This seems to confirm my conclusion.
I have no Pi2 and can't make the test?

Can you help me/us and guide me/us to a solution? (updating the Sense HAT EEPROM or whatever useful)
What is the more low level and efficient way to update an EEPROM ? (required method and tools, from a PC/Linux if necessary)

Thanks

RogerTarani, 2018-08-28:
Since my last post, I did the following at full speed:

1/ flashing the Sense HAT EEPROM, thanks to the RPI-sense site [https://github.com/raspberrypi/rpi-sense](https://github.com/raspberrypi/rpi-sense))
It failed :
>$ make flash
...
 avrdude: error: AVR device not responding 
It looks like the AVR component does not answer (Atmel ATTINY88) .

This was some kind of HW test.

2/ I got a new Sense HAT from a shop.
It displayed rainbow at startup which is a good sign that SHOULD be told on Getting started with Sense HAT.
Then I could have the very first _Hello World!_ example working.

Now, the general question is : 
**How can we fully test the HW of a Pi or a HAT? (get HW state and EEPROM configuration)
Which tools and method are required?**

XECDesign, 2018-08-29:
Out of curiosity, why were you trying to program the EEPROM?

> How can we fully test the HW of a Pi or a HAT? (get HW state and EEPROM configuration)
Which tools and method are required?

You've already done so. The sensors seem okay, but the attiny88 is faulty. It's not responding to avrdude, so there's not much you can do from the pi. The EEPROM may or may not be okay, but there's no reason to look into it, given that we already know that the attiny is faulty.

You can check whether the EEPROM worked by looking at  the output of `sudo vcdbg log msg` or reading it directly with i2c-gpio dt overlay and the eepromutils provided, but that's not something a normal user should need to do.

RogerTarani, 2018-08-29:
Hi Serge, 
I did not intend to program the EEPROM. I just intended to update it, assuming it was required. 

You're right. I did the HW check...thanks to your code. 
At this level, we can decide wether a component is Ok or not then take a new board with a real technical argument for external/internal providers. 

I asked the shop to give me a feed back from the manufacturer to get additional information about that HW. 
We'll see. 
I did manipulate it with care regarding ESD. And the new one was/is fine with the same care. 

Regarding work with students with Sense HAT , I heard that a very few units where not working. 

Thanks 
Best regards

XECDesign, 2018-08-31:
I think the question has been answered anyway.

JamesH65, 2018-08-31:
OK, will close.

closed 2018-08-31 by JamesH65