raspberrypi/documentation #976: Booting Raspberry Pi 3 from mass storage

getz53, 2018-09-02:
The documentation  at
https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md
does not state whether the SD card still needs to be inserted after the OTP bit is set. The Pi does not seem to boot from the USB mass storage device if the SD card is not present. Is this the expected behavior?

getz53, 2018-09-02:
After further research it appears that the SD card should not be needed. The Pi must still be booting from the SD card. I will look for other solutions (e.g., delaying the boot from USB so the disk has time to spin up) and boot with the SD removed.

dgoadby, 2018-09-02:
No, the SD card is not required if you got the configuration right. I use SSD's for booting using a USB3 adaptor with no problems. That said, I did have issues with some HD/USB adaptors which just didn't work at al. I found that USB3 adaptors that use the ASMEDIA chipset work best. The problem seems to be related to the time the USB/Disk adaptor is ready after power up. The on-board LEDS's and adapter LEDS often provide a clue. My SSD's are formatted with a 100 MB FAT partition (/dev/sda1) for the boot and EXT4 for the root partition (/dev/sda2). And finally, I prepared the SSD on the RPi using parted _not_ a Linux PC. I don't know why but, disks prepared on other PC's don't' always work whereas disks prepared on a RPi work fine. I hope some of this helps.

JamesH65, 2018-09-12:
Made some clarification in the USB boot docs. #982

JamesH65, 2018-10-01:
Fix merged.

closed 2018-10-01 by JamesH65