raspberrypi/documentation #978: Add more programming languages to the "Programming with GPIO" section

mekane, 2018-09-04:
On this page: https://www.raspberrypi.org/documentation/usage/gpio/

Only Scratch and Python are mentioned, but the Raspberry Pi comes with more programming languages. It would be nice to have links to programming guides for additional languages.

Specifically JavaScript (Node) and Java would be good to add.

bennuttall, 2018-09-06:
We don't tend to document things that aren't installed by default. And the usage guides are supposed to be a starting point for beginners, so Scratch and Python should be enough.

Perhaps somewhere else we could document or link to things like this. Not sure GPIO, I guess, but it's hard to know where to stop.

mekane, 2018-09-06:
Sure, that makes sense. The kit I bought came with NOOBS on a SD card and has Node 4.8, which is why I went looking for documentation of how to do GPIO in Node. So I came into this thinking that Node was a standard include and should be documented. I understand both the reasons for not including Node by default (out of date Debian repo) and not including external links in the official docs.

I don't totally agree that "Scratch and Python should be enough" for beginners though. JavaScript is a super popular language so if it's possible to program the Raspberry Pi using JS beginners should at least be aware of it I think.

JamesH65, 2018-09-12:
If anyone can provide links to appropriate guides I am happy to add links, but I don't think we have the internal experience or indeed time to write guides for any languages that we don't generally use.