raspberrypi/documentation #985: iptables initialization error

fsofras, 2018-09-17:
While trying to configure the pi as an AP using the documentation from [(https://github.com/SurferTim/documentation/blob/6bc583965254fa292a470990c40b145f553f6b34/configuration/wireless/access-point.md#setting-up-a-raspberry-pi-as-an-access-point-in-a-standalone-network)] , I get the following error:

>  iptables v1.6.0: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded

How can this be fixed?

JamesH65, 2018-10-29:
That is not our documentation repo. Do you get the same problems when using our instructions?

aallan, 2020-06-03:
Waiting for external comment since 2018. Can we close?

closed 2020-06-03 by JamesH65