raspberrypi/documentation #991: 2 Cams 1 Display not working

MOscity, 2018-10-03:
Hello together,

I try to attach 2 SpyCams and an original 7" Raspberry Pi TouchScreen to a Compute Module 3 V1.0 - I have 2 Compute Modules 3, a Compute Module IO Extension Board V3.0 and a Compute Module IO Extension Board Plus. 
But obviously it doesn't work, so therefore I write to you (after 8 hours working on it!).

I followed the guides here:
[Raspi Doc ComputeModule](https://www.raspberrypi.org/documentation/hardware/computemodule/cmio-display.md)

[Video of CJA, who refers also to the guide above](https://www.youtube.com/watch?v=h7sCZs6tWH8
)

and I also studied the dt-blob.bin files in this repository.

First, I wondered why the Pin 2+3 changed to 4+5? [see here](https://github.com/raspberrypi/documentation/blob/92524cce07d65d16cde7799cbb6862b3fa2b405b/hardware/computemodule/dt-blob-disp1-cam2.dts#L1327)
Is there a specific reason to do this?

Anyway, I did as the guide says to connect the GPIO Pins (4+5 to CAM1_IO1 and CAM1_IO0 and all the rest) and attached the TouchScreen to DISP1, but the TouchScreen didn't show anything. I used the 5V + GND / supply directly from the compute module.

When I removed the Display and left the two cams attached (and nothing changed in the GPIO-configuration or wiring), then the 2 cams were recognized and I could use them both (with raspivid in command or with pygame.camera). When I reattach the display, only 1 cam is available (CAM0) and the display doesnt work.

So next I thought why not use Pin 2+3 again, so I changed the corresponding four lines in the dt-blob-disp1-cam2.dts file 

`pin@p2  { function = "output";  termination = "no_pulling"; }; // CAMERA 0 LED`
`pin@p3 { function = "output"; termination = "no_pulling"; }; // CAMERA 0 SHUTDOWN`
`            pin_define@CAMERA_0_LED {
               type = "internal";
               number = <2>;
            };`
`pin_define@CAMERA_0_SHUTDOWN {
               type = "internal";
               number = <3>;
};`

and reattached the wires and the status was the same as before. So I still wonder why you used Pin4+5?
Status AT THE MOMENT:
2 Cams only: both cams work. HDMI Monitor works.
2 Cams + TouchDisplay: only 1 cam works. HDMI Monitor works.

Now some questions:

- Does it has to be that DISPLAY_SDA, DISPLAY_SCL and DISCPLAY_I2C_PORT have the numbers <0>, <1> and <0> ?
            `pin_define@DISPLAY_SDA {
               type = "internal";
               number = <0>;
            };`
`
            pin_define@DISPLAY_SCL {
               type = "internal";
               number = <1>;
            };`
`
            pin_define@DISPLAY_I2C_PORT {
               type="internal";
               number=<0>;
};`
- Does it really has to be "DISPLAY_SDA" and "DISPLAY_SCL"? Because the Cameras use "CAMERA_0_SCL_PIN" (with that "PIN" at the end) ?

- Also I wondered, since there are 2 Ports for 2 Displays (DISP0 and DISP1), why is it called "DISPLAY_SDA" and not for example "DISPLAY_0_SDA" and "DISPLAY_1_SDA" ?

- The lines above come first in the dts file. Are they accidentely being overwritten by this? Because they use the same number. Honestly, I didn't get what this number means yet, is it the GPIO Pin?
`pin_define@CAMERA_0_SDA_PIN {
               type = "internal";
               number = <0>;
            };`
`pin_define@CAMERA_0_SCL_PIN {
               type = "internal";
               number = <1>;
};`

- Another question. I observed that these two lines say both "input" (see below). When I type "sudo raspi-gpio get 46-47" in terminal, then it says that 47 is an output (even after several reboots). Is this normal or did something go wrong in my case?
`pin@p46 { function = "input";   termination = "pull_up";    }; // SMPS_SCL`
`pin@p47 { function = "input"; termination = "pull_up"; }; // SMPS_SDA`

- I can use an external PC-Monitor (with extra supply and HDMI-cable) with 2 Cams attached. In this configuration it is working fine (both cams working, I see something on a monitor). Is it possible that this HDMI connection is always used and it doesn't try to send data to a Display (TouchScreen on DISP1)? (maybe something in the config file (sudo nano /boot/config.txt) )

I really wonder why it is working in that video (second link) and my doesn't work.
I have also a Raspberry Pi 3+, where the TouchDisplay works well - so the TouchDisplay can't be broken.

I would be very glad for some help. This is a serious project and I really need to get this to work.
If you need more information (or pictures?) from me, please tell me.

Thank you very much!

Regards
Matthias

JamesH65, 2018-10-03:
This is not a documentation issue, the best place to ask this sort of question is on the forums, or if you have found a bug, in the userland or linux kernel issue tracker (depending on where the bug is)

closed 2018-10-03 by JamesH65

MOscity, 2018-10-14:
Found the solution, see here: https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=123896&start=25

The problem was that it is not clearly stated in the documentation, that it is a special Display Adaptor for the compute module, which is switching the pin routing. I thought I could use a normal CSI/DVI Extender (Pin-to-Pin) with a Raspberry Zero Cable.

6by9 will create an Issue for the docs on this matter.