raspberrypi/documentation #997: net_tutorial.md under-specifies the files needed in the tftproot

zbrozek, 2018-10-14:
The documentation claims that only bootcode.bin and start.elf are required, but this is incorrect. When booting many more files are requested - some which exist and some which do not - but if you haven't copied everything you haven't satisfied the requirements.

The documentation should include the complete manifest of required files, or clearly state that all of the boot folder is required.

JamesH65, 2018-10-15:
Can you provide a link to the documentation you believe is incorrect please?

zbrozek, 2018-10-15:
Certainly!

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md

JamesH65, 2018-10-15:
fixed by #999

JamesH65, 2018-10-29:
Closing this issue as questions answered/issue resolved.

closed 2018-10-29 by JamesH65