raspberrypi/documentation #1006: documentation/hardware/raspberrypi/bootmodes/net_tutorial.md assumes content in /etc/network/interfaces

petemoore, 2018-10-19:
This tutorial asks the user to change the line `iface eth0 inet manual` in `/etc/network/interfaces` but this line no longer exists in raspbian release.

I found [this guide](http://sizious.com/2015/08/28/setting-a-static-ip-on-raspberry-pi-on-raspbian-20150505/) to help me, not sure if it is the correct approach or not.

rbisewski, 2018-10-21:
Hmmm... seems there are several things in that tutorial that are a bit out-of-date.

The `interfaces` file was used more so in the past when `ifup` was the preferred method of adjusting network cards. This is no longer the case. Consider using either `NetworkManager` or if you are attempting to run DHCP server maybe do:

```
sudo vim /etc/dhcpcd.conf
```

Which I think should have the following commented lines at the bottom:

```
# Example static IP configuration:
#interface eth0
#static ip_address=192.168.0.10/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.0.1
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
```

Uncomment the necessary lines and fill them with information about your network card.

Likewise with directly editting `/etc/resolv.conf`, I believe this has been superseded by systemd's own configuration. If you need to add a name server consider doing this instead:

```
sudo vim /etc/systemd/resolved.conf
```

Then adjust the name server to these lines:

```
[Resolve]
#DNS=
#FallbackDNS=
```

Feel free to let me know if you stuck on at any point.

JamesH65, 2019-01-18:
@rbisewski HI, you seem to know about all this stuff, would you be able to provide a PR on the page in question with the updated information?

rbisewski, 2019-01-20:
@JamesH65 I have since created the pull request for the tutorial as discussed. Admittedly I normally use the Ubuntu 18.04 image found [here](http://cdimage.ubuntu.com/releases/18.04/beta/) for my personal server and not the [official Raspbian image](https://www.raspberrypi.org/downloads/raspbian/).

Let me know what you think, and feel free to clean it up and merge if you want.

lurch, 2019-02-06:
> Admittedly I normally use the Ubuntu 18.04 image found [here](http://cdimage.ubuntu.com/releases/18.04/beta/) for my personal server and not the [official Raspbian image](https://www.raspberrypi.org/downloads/raspbian/).

Just for clarity, I believe the Raspberry Pi documentation is targeted entirely at Raspbian as the supported OS, I think support for any other OSes are up to 3rd parties to provide?

JamesH65, 2019-02-06:
> Just for clarity, I believe the Raspberry Pi documentation is targeted entirely at Raspbian as the supported OS, I think support for any other OSes are up to 3rd parties to provide?

Indeed. If you are making changes to the docs, they MUST work on Raspbian.

JamesH65, 2020-11-19:
This whole area of docs has changed a lot since this PR. So closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2020-11-19 by JamesH65