raspberrypi/documentation #1008: Authentication failed during client connection to Raspberry Standalone Access Point

LaurentC53, 2018-10-21:
Hi,
I have set up a Raspberry Pi Zero (with Raspbian Stretch) as a standalone access point, following the official guideline (https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md). Everything works fine (thanks a lot !!) except one point :
Despite my clients (windows 10 laptop or Android 5.0 tablet) have been configured to connect to the Access Point automatically, after Raspberry Pi reboot, SSID network is visible but clients wait for a manual action to connect : click on 'connect' for Windows 10 or password entry for Android.
Is that normal ? Or is there any configuration missing (like auto wlan0 ...) ?
Many thanks by advance,
Regards,
LC

JamesH65, 2018-10-29:
Not sure what is normal here. I'll need to read up on it a bit.

JamesH65, 2019-01-18:
@SurferTim You had some input on the docs here, any ideas?

SurferTim, 2019-01-22:
I'm not sure. My devices seem to reconnect even after a reboot. Don't have a Windows 10 or Android device to test with.

aallan, 2020-06-03:
Open for more than a year waiting for external comment. Possibly should close?

closed 2020-06-03 by LaurentC53