raspberrypi/documentation #1018: Failed to start hostapd.service: Unit hostapd.service is masked.

ijash, 2018-10-30:
`sudo systemctl start hostapd `returned `Failed to start hostapd.service: Unit hostapd.service is masked.`

JamesH65, 2018-10-30:
I have no idea why you have posted this issue. Do you have a specific part of the documentation that is showing this error?

ijash, 2018-10-31:
> I have no idea why you have posted this issue. Do you have a specific part of the documentation that is showing this error?
sorry for not pointing the documentation.

i have followed https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md
and get those error during `sudo systemctl start`.

i am using RPi3 B

ijash, 2019-01-18:
i'll close this topic, as i long forgot the steps to reproduce.

beta-tester, 2019-03-10:
i have the same issue.

today, i took the actual rasbian image [2018-11-13-raspbian-stretch.zip](https://downloads.raspberrypi.org/raspbian/images/raspbian-2018-11-15/2018-11-13-raspbian-stretch.zip)
and flashed it to an SD card for my RPi3B+.

i enabled SSH by adding a `ssh` file to boot partition befor very first boot.

at reboot (after automatic resizing/growing filesystem pass) i could shortly see a message with something like that the WiFi was disables because of unsetted country code or something similar.

from SSH i ran `sudo raspi-config` to setup **I4: WiFi country code** to DE for Germany.

after `sudo apt update && sudo apt upgrade -y` i installed packages `sudo apt install -y dnsmasq hostapd` to be able to setup Wifi AccessPoint i tried to setup my RPi3B+ folowed the steps for [Access-Point (NAT)](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md)

but when i try to start/enable hostapd with `sudo systemctl start hostapd` i get an error message
`Failed to start hostapd.service: Unit hostapd.service is masked`.

how can i enable WiFi again? how can i unmask hostapd?

XECDesign, 2019-03-10:
@beta-tester try unmasking, enabling and starting the service:
https://github.com/raspberrypi/documentation/pull/1097/files#diff-bab12433eaae7aea98b21a2978c8ba52R223

beta-tester, 2019-03-10:
yes, thank you this is working.
```
sudo systemctl unmask hostapd
sudo systemctl enable hostapd
sudo systemctl start hostapd
```

scudrunner, 2019-03-11:
THANK YOU!  I have been having the same issue with fresh installs, and the three lines above seem to have solved it!

ijash, 2019-03-11:
> @beta-tester try unmasking, enabling and starting the service:
> https://github.com/raspberrypi/documentation/pull/1097/files#diff-bab12433eaae7aea98b21a2978c8ba52R223

shouldn't this be on the documentation page?

XECDesign, 2019-03-11:
> shouldn't this be on the documentation page?

It's just the behaviour of the package that was backported from buster. I'm guessing the logic is that if hostapd is installed, it should be enabled explicitly by the user. The postinst script is checking whether the conf file exists before masking it, but I'm not sure if that's working as intended. I've only run through the instructions from a clean start and haven't tested updating from a previously working hostapd configuration.

> shouldn't this be on the documentation page?

It's undergoing review and copy editing, which may take a while.

JamesH65, 2019-03-11:
With regard to updating the docs, @XECDesign has provided a PR, but unfortunately, staff shortages in our copy edit department means there is a long backlog to get things approved. I've asked for this to be given priority.

heartlandcoop-prichards, 2019-03-13:
Thank you, I was having the same problem.  I believe this issue just appeared in the last few weeks.

bmoren, 2019-03-14:
checking in on this issue! it was masked for me...

ijash, 2019-03-14:
Should i reopen the issue? Even though the fix is found, just letting the others found this case easily before the documentation is being updated..

XECDesign, 2019-03-14:
Not sure. We already have #1093, #1095 and probably a few others.

bmoren, 2019-03-14:
it seems logical to me to re-open. given 1. that the name of this issue is exactly the error you get when trying to config the documentation and the solution:
```
sudo systemctl unmask hostapd
sudo systemctl enable hostapd
sudo systemctl start hostapd
```
is presented here nicely by @beta-tester
and 2. that this issue really hasn't been solved yet. eg, the documentation hasn't actually been updated.

beta-tester, 2019-03-14:
> 
> ```
> sudo systemctl unmask hostapd
> sudo systemctl enable hostapd
> sudo systemctl start hostapd
> ```
> 
> is presented here nicely by @beta-tester.

that is not my merit. that solution came from the documentation of @XECDesign

weasdown, 2019-04-02:
@XECDesign's solution didn't work for me. The first two commands ran without any problems, but running `sudo systemctl start hostapd` generated an error: `Job for hostapd.service failed because the control process exited with error code. See "systemctl hostapd.service" and "journalctl -xe" for details.`. 

When I ran `systemctl hostapd.service`, the following output was generated: `● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2019-04-02 12:42:00 BST; 2s ago
  Process: 1557 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE

Apr 02 12:42:00 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited status=1
Apr 02 12:42:00 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Apr 02 12:42:00 raspberrypi systemd[1]: hostapd.service: Unit entered failed state.
Apr 02 12:42:00 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Apr 02 12:42:02 raspberrypi systemd[1]: hostapd.service: Service hold-off time over, scheduling restart.
Apr 02 12:42:02 raspberrypi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Apr 02 12:42:02 raspberrypi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...`

XECDesign, 2019-04-02:
Your issue is not related to this one.

Is this using the onboard wifi? Can you post your hostapd config file?

weasdown, 2019-04-03:
I thought the issue was related because I was having the same issue as at the start of the thread while using the same Wi-Fi hotspot tutorial. I am using the onboard Wi-Fi on a Pi 3B+ with a clean install of the 2018-11-13 version of Raspbian and updates installed as recommended on first boot. Here's the hostapd.conf file:
[hostapd.txt](https://github.com/raspberrypi/documentation/files/3038425/hostapd.txt)

XECDesign, 2019-04-03:
Does it work if you use hw_mode=g instead? That's the only difference I see.

weasdown, 2019-04-03:
It does work with hw_mode=g, strangely. Below are versions of the log files with the two levels of verbosity. I have also noticed that if I set it to hw_mode=a and try starting it, the taskbar Wi-Fi icon gets stuck in a loop, going up to two bars then instantly cutting to the double x icon, briefly showing the two bars before starting the loop again.
[hostapd_d.log](https://github.com/raspberrypi/documentation/files/3039350/hostapd_d.log)
[hostapd_dd.log](https://github.com/raspberrypi/documentation/files/3039351/hostapd_dd.log)

XECDesign, 2019-04-03:
> 1554303941.345443: 1554303941.345448: wlan0: IEEE 802.11 Configured channel (7) not found from the channel list of current mode (2) IEEE 802.11a

Channel 7 isn't a valid 5GHz channel, so that makes sense.

weasdown, 2019-04-03:
I'm in Europe, so tried channel 36, and that works fine - thank you! I tried channel 38, 50 and 106 out of curiosity too though and they fail with that same error as before about the channel not being in the list
[hostapd_38d.log](https://github.com/raspberrypi/documentation/files/3039477/hostapd_38d.log)
[Uploading hostapd_50d.log…]()
[Uploading hostapd_106d.log…]()

. Here are the logs for all three cases:

XECDesign, 2019-04-03:
The hardware is certified for the North American region, so I suspect it filters out some channels which may be acceptable in your region. It's hard to get any clear answers about anything when it comes to 5GHz regulatory domains.

weasdown, 2019-04-03:
Interesting, thanks. If I need to use 5GHz then I'll use channel 36, but for my application 2.4GHz should be more than adequate, so I'll just use that.

XECDesign, 2019-04-03:
iw list shows the frequencies that are available. Then select whatever is appropriate for your region from that subset.

weasdown, 2019-04-03:
Great, thanks for the help.

XECDesign, 2019-04-03:
np

beta-tester, 2019-04-03:
> 
> 
> I'm in Europe, so tried channel 36, and that works fine - thank you! I tried channel 38, 50 and 106 out of curiosity too though and they fail with that same error as before about the channel not being in the list. Here are the logs for all three cases:
> [hostapd_38d.log](https://github.com/raspberrypi/documentation/files/3039487/hostapd_38d.log)
> [hostapd_50d.log](https://github.com/raspberrypi/documentation/files/3039488/hostapd_50d.log)
> [hostapd_106d.log](https://github.com/raspberrypi/documentation/files/3039489/hostapd_106d.log)

in my config file i have `country_code=DE` for germany, to setup the region.

```
# Country code (ISO/IEC 3166-1). Used to set regulatory domain.
# Set as needed to indicate country in which device is operating.
# This can limit available channels and transmit power.
# These two octets are used as the first two octets of the Country String
# (dot11CountryString)
#country_code=US

```

weasdown, 2019-04-03:
What does this do? I haven't noticed any difference in the output from `iw list`.

beta-tester, 2019-04-04:
i expect that this option is regulating the available channels and the transmitting power of the wifi for the selected region...
i didn't test it, i only used what is written in the description [here](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf).

Doyled96, 2019-04-13:
I am still having problems with this. I am located in the US followed the directions exactly and hit the same speed bumps with hostapd being masked. Now facing the issue with the error code. PLEASE HELP!

JamesH65, 2019-04-15:
This really isn't a documentation issue, so should really be posted in the linux issue tracker where it will get more views. You could also try the forums.

GitHubIsToxic, 2019-04-18:
Just a heads-up: I spend a few hours trying to solve this issue...and it turned out my password phrase in "hostapd.conf" was too short. As result hostapd doesn't start, apparently leading to the 'mask' thing issue.

Yunir, 2019-12-01:
@GitHubIsToxic, had the same issue with password size - thanks a lot! We should notice that our passwords should be greater than ...

vldp2000, 2020-03-10:
that is crazy. It has to tell that something is wrong with the password instead of giving the meaningless errors

closed 2021-06-17 by aallan

Splines, 2022-04-21:
The [Raspi documentation](https://www.raspberrypi.com/documentation/computers/configuration.html#ap-config) is pretty clear about this:

> Note that the name and password should not have quotes around them. The passphrase should be between 8 and 64 characters in length.

But I definitely agree with @vldp2000 that this should have its own error message. Note that the [hostapd conf](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf) states:

```
# SAE password
# This parameter can be used to set passwords for SAE. By default, the
# wpa_passphrase value is used if this separate parameter is not used, but
# wpa_passphrase follows the WPA-PSK constraints (8..63 characters) even though
# SAE passwords do not have such constraints. If the BSS enabled both SAE and
# WPA-PSK and both values are set, SAE uses the sae_password values and WPA-PSK
# uses the wpa_passphrase value.
```

So this is where this constraint comes from. I've just reported this issue directly to the developer of hostapd Jouni Malinen <j@w1.fi> (see `hostapd -v` for this kind of information). By the way, the hostapd repository can be found [here](https://w1.fi/cgit/hostap/tree/hostapd).