raspberrypi/documentation #1029: documentation/msd.md incorrectly states only Pi 3 can use USB MSD boot

andrum99, 2018-11-27:
The official documentation at documentation/msd.md incorrectly states only Pi 3 can boot from a USB mass storage device. I'm guessing it just hasn't been updated since other 2837-based Pi's came out. The fix is to explain that the Pi 2 B Model 1.2, Pi 3 B and Pi 3 B+ can do MSD boot, but the rest of the range cannot.

Also bear in mind that MSD boot is currently blocked on the 2837-based Pi 3 A+, since the OTP bits are set such that it goes into USB host mode.

JamesH65, 2018-11-27:
Are you referrring to hardware/raspberrypi/bootmodes/msd.md? Because that refers to both the 3 and 3+, the only bit missing is the v1.2 Model 2 I believe?

JamesH65, 2018-11-27:
Ah, I see you have sentin a PR, I'll move to that.

andrum99, 2018-11-27:
Sorry - I got the address of the page wrong - github obviously shortened it. The page I'm referring to is documentation/hardware/raspberrypi/bootmodes/msd.md. It is directly linked to from documentation/faqs/README.md - that page has no mention of the fact that only some Pi models support MSD booting.

Yes PR is #1030 

HTH

JamesH65, 2019-01-07:
Merged so closing.

closed 2019-01-07 by JamesH65