raspberrypi/documentation #1040: Typo in introduction

em-er-es, 2018-12-08:


lurch, 2018-12-10:
"Release 2, Page 5" of what exactly? Could you point to a URL?

em-er-es, 2018-12-11:
The datasheet for Raspberry Pi Compute Modules, release 2: [rpi_DATA_CM_2p0.pdf](https://www.raspberrypi.org/documentation/hardware/computemodule/datasheets/rpi_DATA_CM_2p0.pdf) from [CM documentation page](https://www.raspberrypi.org/documentation/hardware/computemodule/datasheet.md).

JamesH65, 2018-12-11:
OK, can see the issue, will need to find out who is responsible for that document - it's not part of the easily fixed docs repo in github.

@jimbojr This is for you I think.

lurch, 2019-01-09:
I just spotted that Table 4 on Page 13 also says "GPIO28-27 I/O Supply Voltage" which I suspect should be "GPIO28-45 I/O Supply Voltage" :slightly_smiling_face:

JamesH65, 2020-11-19:
@dp111 Are you now the goto guy for CM (older ones) datasheet mods? Is this error still present?

dp111, 2020-11-19:
@JamesH65  not me I'm afraid for older datasheets

lurch, 2020-11-19:
Probably still @jimbojr then.

aallan, 2021-06-07:
It's probably my job to figure out who has the source text for these and then take ownership.

aallan, 2021-08-13:
Fixed.

closed 2021-08-13 by aallan