raspberrypi/documentation #1042: Stop recommending Etcher for image writing

aladar42, 2018-12-12:
According to your documentation, I used Etcher to write the image, but it only resulted in completely corrupting TWO brand new SD cards. I looked around and its not an isolated issue by any means:
https://github.com/balena-io/etcher/issues/1426
https://github.com/balena-io/etcher/issues/2532
https://raspberrypi.stackexchange.com/questions/72367/sd-card-corrupts-every-time-after-writing-image-on-windows-using-etcher
My suggestion is to not recommend Etcher as a first thing on your site, or at least display very clear warnings that it may corrupt SD cards. I am now $20 in the hole because of this, and I believe others shouldnt only learn about the softwares faults after their cards get rendered useless. Please consider this.

lurch, 2018-12-12:
ping @zvin @thundron @jviotti

JamesH65, 2018-12-12:
Rather than unrecommending Etcher (which I have never had any problems with whatsoever, this is the first report I have seen on any sort of issues), I think this needs to go to the Etcher devs (I presume these are the people @lurch has pinged) for fixing.

Etcher does work for the vast majority of people, is multi platform, and much easier to use than the alternatives, which is why we moved to recommending it in the first place.

lurch, 2018-12-12:
@JamesH65 Yes, AFAIK those are the current Etcher developers (sorry for not mentioning that).

@aladar42 See [this comment](https://github.com/balena-io/etcher/issues/1426#issuecomment-421938957) in particular.

jviotti, 2018-12-12:
@aladar42 Please see https://github.com/balena-io/etcher/issues/1426#issuecomment-421938957. We had similar comments in the past and we are working on UX improvements to make it more obvious but just to be clear: Etcher does not corrupt SD Cards nor even has the capability to do so, given that the device is guarded by the operating system anyways. All interactions Etcher has with the devices happen through the OS system calls, which as far as my knowledge goes, wouldn't even permit something like this (what would be the point of an operating system otherwise?)

The most common case is that users flash images that contain file-systems that their host operating systems can't read (i.e. flashing a GNU/Linux image and trying to read it back on Windows). In that case, the host operating system complains that it can't read the device (rightfully) and might even struggle to format it using the traditional solutions. From my own experience, the Windows formatting tool for example, fails in almost all cases that are not trivial. I had a lot of luck wiping devices using `diskpart`, which seems to be much more robust.

@aladar42 Please try the methods I outlined in the linked comment, or even try to read the SD Card on a computer running an operating system that can read those file-systems out of the box. Chances that your SD Card is actually corrupted is pretty slim, and even if that's the case, its most likely to be caused by faulty hardware (i.e. bad quality SD Card reader) than by something like Etcher.

aladar42, 2018-12-13:
I have read those comments already, however they do not change anything. When I write the image with Etcher, the card always ends up corrupted. The pattern is always the same - first write on brand new card (from a trusted retailer, no eBay, no Alibaba, fresh from packaging) usually writes the image, but RPi does not boot, and subsequent writes on the card are unsuccesful as every tool, including dd hangs during writing. Reformatting the drive using gparted or any other tool usually goes well, but no RPi system can ever be written to it again. badblocks reports millions of read failures. This repeated 3 times now over 3 different cards and different card readers. There definitely is a pattern of failure with Etcher, and if not unrecommended, users should at least be warned in some form on the official RPi site.

JamesH65, 2018-12-13:
This is bordering on weirdly unbelieveable. I've personally used Etcher dozens of times and never had a problem, thousands (10's of thousands?) of our users use Etcher all the time, with no problems. So this is almost certainly related to the system you are using and some weird interactions. So, I see you have tried different adapters, but was it the same PC/laptop each time? Windows or Linux, or Mac? Really asking - what were the similarities for each failure? What make are the SD cards?

aladar42, 2018-12-13:
Two different PCs. One running Debian, the other running Arch. The SD cards were Kingston and Adata. The images flashed were LibreELEC, DietPi, Arch ARM and Raspbian. As I said, its obviously not corrupting it physically, the test results clearly show difference between "properly" corrupted card (which is showing corrupted blocks and faulty writes using badblocks, vs this one only showing errors reading superblocks) and a card that has been corrupted by Etcher, but in the end its still not useable in RPi after flashing. I wouldnt think much of it if there wasnt so many other people reporting the same issue.

zvin, 2018-12-13:
@aladar42 
You can not break an SD card with some software.

JamesH65, 2018-12-13:
TBH, after seeing this issue and doing a search, the number of people seeing the issues is probably < 20, and given the huge number of people using Etcher, that's pretty insignificant. Interesting that you are using Linux, because most of the issues I read were WIndows based. Since the underlying process of writing to the card is dependent on the OS,  this makes things even weirder.

aladar42, 2018-12-13:
Thats why I am requesting that you at least add a notifier, or something like that. Im sure those 20 or so people dont really appreciate having their money wasted on cards they cant now use. I went with Etcher because I didnt remember the dd command from the top off my head, and I thought that if RPi Foundation is recommending it, it is probably the best method to go about it, only to later find out that I am definitely not alone who got hit by this. If I knew this or read about it beforehand, I would just use dd or booted into Windows for some other software, because I cant really afford to buy multiple cards one after another. Its ultimately up to you, but I for one would definitely appreciate the heads up before the corruption, not after the thing is done...

thundron, 2018-12-13:
@aladar42 Could you share results when using `dd` once you restored your cards?
You can use the command listed [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#gnulinux) to write zeroes to the sdcard then formatting it properly with your OS's tools if needed

thundron, 2018-12-13:
@aladar42 About the superblocks, that's actually something done by whoever makes the image to make them boot on many systems, for reference see [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos)

ssokolow, 2018-12-17:
Technically, it *is* possible for software to ruin an SD card if there's a defect in the SD card's memory controller.

For example, a bunch of the negative reviews from around 2014 for [this 8GB Kingston MicroSD card](https://www.newegg.ca/Product/Product.aspx?reviews=all&Item=9SIA66K3114699&SortField=0&SummaryType=0&PageSize=36&SelectedRating=-1&VideoOnlyMark=False&ignorebbr=1&IsFeedbackTab=true#scrollFullInfo) received the following response:

> [...] We apologize for the issue experienced with the microSDHC cards. We stand by all our products and can replace those cards under their lifetime warranty. If they were used in Android 4.0 or later devices, the issue may be related to a problem that was discovered last year that caused the cards to become corrupt and then undetectable in any device. This issue was resolved August 2013 but it is possible you may have received cards that were manufacturer prior to that time. [...]

zvin, 2018-12-17:
@ssokolow this is what I would call a defective sd card.

ssokolow, 2018-12-17:
Certainly... it's just that, normally, when people think "defective", they think a transient manufacturing defect that QC failed to catch, not a bug in the firmware which consistently affects entire manufacturing runs.

JamesH65, 2019-01-07:
Since we are not going to "unrecommend" Etcher, and the number of people seeing issues is tiny comapred with the number of users, I think this can be closed.

closed 2019-01-07 by JamesH65