raspberrypi/documentation #1044: Simpler way to verify correct image writing on Linux

ssokolow, 2018-12-17:
The instructions for verifying a correct write in `linux.md` are needlessly complex.

This will do just fine:

1. Take a hash of the image before writing (I used `sha256sum` but `sha1sum` or `md5sum` will do just fine for this purpose):

        $ sha256sum /tmp/2018-11-13-raspbian-stretch-lite.img                
        b9e22d7592c6936e4b0adadc9c9dbf7b4868cbfc26ff68a329e32f4af54fed70  /tmp/2018-11-13-raspbian-stretch-lite.img

2. Get the size of the image in bytes (I used `du -b` for simple output, but `ls -l` works too)

        $ du -b /tmp/2018-11-13-raspbian-stretch-lite.img
        1866465280      /tmp/2018-11-13-raspbian-stretch-lite.img

3. Get a hash of that many bytes off the beginning of the SD card (I use `sudo sh -c '...'` so that any output from sudo won't get included in what's being hashed)

        $ sudo sh -c 'head -c1866465280 /dev/sdi | sha256sum'
        b9e22d7592c6936e4b0adadc9c9dbf7b4868cbfc26ff68a329e32f4af54fed70  -

4. Compare the two hashes.

        $ python
        Python 2.7.12 (default, Jul 17 2016, 01:21:14) 
        [GCC 4.8.4] on linux2
        Type "help", "copyright", "credits" or "license" for more information.
        >>> 'b9e22d7592c6936e4b0adadc9c9dbf7b4868cbfc26ff68a329e32f4af54fed70' == 'b9e22d7592c6936e4b0adadc9c9dbf7b4868cbfc26ff68a329e32f4af54fed70'
        True
        >>> (Ctrl+D here)

If you're feeling lazy, just verify that the first and last 3-4 characters match. These kinds of hashes are designed to change drastically and unpredictably in response to small changes, so the odds of a corrupted write leaving matching sequences at the beginning and end but changing something in the middle are miniscule.

lurch, 2018-12-17:
Using `head` to read a specific number of bytes from the SD card is a clever idea. I guess you could also do:
```
$ sudo sh -c "head -c $(stat -c '%s' /tmp/2018-11-13-raspbian-stretch-lite.img) /dev/sdi | sha256sum"
b9e22d7592c6936e4b0adadc9c9dbf7b4868cbfc26ff68a329e32f4af54fed70  -
```
to combine steps 2 and 3? :wink:

ssokolow, 2018-12-17:
You could, but I was trying to go for a nice balance between conciseness and comprehensibility. 

Wrapping it in `sh -c '...'` feels like a nice place to draw the line.

aallan, 2020-06-03:
Open since 2018. Did this ever get integrated? If so can we close. If not, is it still relevant?

JamesH65, 2020-06-03:
Still relevant I believe, it's certainly an easier process than the current one here:

https://www.raspberrypi.org/documentation/installation/installing-images/linux.md

andrum99, 2021-03-20:
Superceded by Raspberry Pi Imager: this issue can be closed.

closed 2021-06-07 by aallan