raspberrypi/documentation #1049: Raspberry Pi 3 A+ Schematics

nullr0ute, 2018-12-27:
It would be useful to add the reduced schematics, similar to the 3B and the 3B+ to the docs here:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md

It would also be useful to have the pinouts and connections to the WiFi/BT module as part of that, there's been issues/regressions upstream recently and the way that module is connected to the rest of the SoC would be useful to have as public docs. (an updated 3B/3B+/ZeroW with that details would be nice too)

JamesH65, 2018-12-28:
@Roger-Thornton Are these in the pipeline?

lurch, 2018-12-28:
IIRC when the WiFi/BT module was asked about before (or maybe it was something else), somebody pointed to https://github.com/raspberrypi/documentation/blob/master/configuration/pin-configuration.md#sample-device-tree-source-file ?

nullr0ute, 2018-12-29:
The problem with the device tree is that there's been issues with it in the past, which depending on how the code is (built in vs modular) the issues may appear differently, see the recent upstream WiFi regression. There was a regression that wasn't seen on Raspbian because of the way the it deals with it vs a generic distribution, hence my request of actual schematics as to how things are physically connected up.

JamesH65, 2019-02-06:
@Roger-Thornton Are we likely to produce 3A+ schematics at some point?

JamesH65, 2019-02-07:
Uploaded to githuib, should appear here https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md soon.

closed 2019-02-07 by JamesH65