raspberrypi/documentation #1050: raspbian/updating.md: additional information and third-party solutions

mirzak, 2018-12-28:
I have a suggested update of the of the rasbpian [updating.md](https://github.com/raspberrypi/documentation/blob/master/raspbian/updating.md) file and I wanted to get some feedback before trying to send a PR.

I have created a draft which is available [here](https://github.com/mirzak/documentation/commits/updating-raspbian). 

The goal of the update to bring a bit more awareness regarding importance of software updates but also provide some alternatives that are available as open-source projects. Currently I have added mender.io which supports Raspbian out of the box.

I am affiliated with the Mender project, but I also have a history of involvement in software updates on Linux in general and I have written several articles on the topic ([article1](https://mkrak.org/2017/11/18/updating-embedded-linux-devices-part-0/), [article2](https://mkrak.org/2018/01/10/updating-embedded-linux-devices-part1/)) covering several projects.

I am looking forward to get your feedback and would happily work on this if you deem it something of interest to the greater Raspberry Pi community.

JamesH65, 2018-12-28:
@ghollingworth @LizUpton What's our current policy for changes such as this?

ghollingworth, 2019-01-03:
I'm happy pulling this in, agree with the warnings over CVE and the mender link will be useful to industrial users of the Pi.

Gordon

JamesH65, 2019-01-03:
OK, @mirzak can you submit your final draft as a pull request please. I'll review it then pass it on to our copy editing team for final polish, then it can be merged.

mirzak, 2019-01-03:
Thank you for the feedback, I will continue working on this then and you can expect a pull request mid next week.

mirzak, 2019-01-16:
@JamesH65, I created the pull-request as agreed. Wondering if you had any time to look at it?

JamesH65, 2019-01-16:
Thanks for the reminder. Read it, seems OK, now pushed to our copy edit team. Usually takes between one and two weeks to get to the top of their work list, so should go online after that. Thanks!

mirzak, 2019-02-01:
@JamesH65, any update on the progress of copy edit team?

JamesH65, 2019-02-02:
I usually batch up and then submit in a chunk. This one has been submitted and is due by end of next week, as the editors are pretty busy at the moment.

mirzak, 2019-02-28:
Checking if there are any updates?

JamesH65, 2019-02-28:
We've had a lot of illness in the comms team, and our copy editor has moved from Trading to Foundation so having trouble getting the CE's through at the moment. Hopefully get some movement soon.

lurch, 2020-04-29:
Closing, as #1062 has been merged.

closed 2020-04-29 by lurch