raspberrypi/documentation #1058: Apache / PHP installation instructions error

n4mwd, 2019-01-08:
Your page at: https://www.raspberrypi.org/documentation/remote-access/web-server/apache.md

Has incorrect information. The line: "sudo apt-get install php libapache2-mod-php -y"
is incorrect and will not work. It should be: "sudo apt-get install php5 libapache2-mod-php5 -y"

JamesH65, 2019-01-08:
I've just tried this on a 4.19 kernel on Stretch and the install went OK. Ended up installing php7 I beleive. I haven't tested the installation actually works however. What error were you seeing that makes you think this change is necesssary?

```
The following additional packages will be installed:
  apache2 apache2-bin apache2-data apache2-utils libapache2-mod-php7.0 libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap liblua5.2-0 php-common php7.0 php7.0-cli php7.0-common php7.0-json php7.0-opcache php7.0-readline
  ssl-cert
Suggested packages:
  apache2-doc apache2-suexec-pristine | apache2-suexec-custom php-pear openssl-blacklist
The following NEW packages will be installed:
  apache2 apache2-bin apache2-data apache2-utils libapache2-mod-php libapache2-mod-php7.0 libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap liblua5.2-0 php php-common php7.0 php7.0-cli php7.0-common php7.0-json php7.0-opcache
  php7.0-readline ssl-cert
0 upgraded, 20 newly installed, 0 to remove and 84 not upgraded.
 ```

bennuttall, 2019-01-08:
The documentation is for Raspbian Stretch, where the package `php` is PHP7, and there is no PHP 5.

n4mwd, 2019-01-08:
I submitted that to Raspbian documentation a while back so I don't remember the exact error.  They said to post it here.  I was installing on a RPi3B+ that had never had apache or Php.  Apache installed fine, but PHP gave several errors.  I don't remember exactly, but it was something like:

"Reading state information... Done
E: Unable to locate package php
E: Unable to locate package libapache2-mod-php
E: Unable to locate package php-mcrypt"

Changing php to php5 worked.

n4mwd, 2019-01-08:
I'm not a Linux expert so I'm not sure if I'm running Stretch or not.  Logging in gives the following version:
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l

This is a documentation error in that its doesn't say my version requires php5 instead of php.

bennuttall, 2019-01-08:
`lsb_release -a` gives information about your distribution. `lsb_release -sc` gives the shortcode i.e. `stretch` or `jessie`. We don't support Jessie any more so all the documentation is for Stretch, and we've checked this is correct.

lurch, 2019-01-08:
Looks like a duplicate of #796

bennuttall, 2019-01-08:
It's the _opposite_ of #796...

n4mwd, 2019-01-08:
I ran the program and it said I was using Jesse.  I got to the page in the OP via google and there was nothing I could see about Jesse being unsupported.  Not sure where the Jesse/Apache documentation is.  Thanks for clarifying this.

closed 2019-01-08 by n4mwd

lurch, 2019-01-08:
I'm not sure if there's anywhere where it explicitly says that Jessie is unsupported ( @JamesH65 ? ) but basically the Raspberry Pi Foundation only support the [latest release](https://www.raspberrypi.org/downloads/raspbian/) of Raspbian, which is currently Stretch.

n4mwd, 2019-01-08:
Yes it does look like #796.  According to a comment there, the code was changed in Stretch so that now only 'php' is required.  I assume that 'php' is now a package that redirects to 'php7' or whatever the latest is..