raspberrypi/documentation #1059: CAD files (3d) for designing 3D printing, parts, etc

jgwinner, 2019-01-08:
Hello:

My understanding from a few files on Thingiverse, is that there were some design files for the various Raspberry Pi physical boards.

I guess manufacturers could have some components slightly different in size than others, but I think more than a DXF diagram, a reference implementation / official guidelines for things like volume, intrusions, etc. would be really handy; preferably on the mechanical drawings page.

Forgive me if this information is posted somewhere else; I just checked and only saw PDF and DXF file formats.

For people making cases, the "attribution" requirements on a mostly non-commercial site like Thingiverse may be a problem too, making an official 3D file (Blender, OBJ, GLTF, or other open format) a really great thing.

Thank you for listening,

== John ==

JamesH65, 2019-01-10:
Will ask around, but I don't believe we have anything like that at all, and it seems unlikely we would generate them given our workload. Sorry.

lurch, 2019-01-10:
See also #95

JamesH65, 2019-02-07:
I cannot see us doing this in the near or even medium term. So closing as a Won't Fix. This means that although the issue is acknowledged, it is felt that no further action is likely.

closed 2019-02-07 by JamesH65

jgwinner, 2019-02-07:
Darn. Thanks for your consideration.

Some CAD files do it automatically, was hoping it was just an export, but I understand on the workload issue. There are some versions on the web but some are pretty sketchy. For reference, if others stumble on this issue:

https://3dwarehouse.sketchup.com/model/f18fc33e-3c42-433f-aa51-021c22689148/Raspberry-PI-model-B

https://grabcad.com/library/raspberry-pi-model-b-2

I do not know how accurate these are.

        == John ==

JamesH65, 2019-02-07:
We don't have CAD files for the boards, they are done on PCB design tools, not CAD systems, so there is no export possible AFAIK.

jgwinner, 2019-02-07:
It depends on what tool. I know AutoTRAX DEX does it. I think a few others do too, it's been a while since I've used PCB software, frankly.

        == John ==