raspberrypi/documentation #1065: vcgencmd documentation

JamesH65, 2019-01-15:
There don't appear to be any pages dedicated to the vcgencmd helper application, which is often referred in the forums or when trying to help users with issues.

andrum99, 2019-01-15:
Quite a simple command. Is it open source? I think the first step should be adding basic usage info to the binary.

JamesH65, 2019-01-15:
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/gencmd/gencmd.c

But most of the heavy lifting is done on the other side of the link to the videocore, as you might expect! There is scope for adding help information to the ARM end, but the list of commands is sent back from the VC, so when new commands are added they might get missed from adding to the help at this end.

andrum99, 2019-01-15:
I see. Is there a command you can give vcgencmd that will display all of
the commands?

andrum99, 2019-01-15:
You will of course have exactly the same problem with any documentation for this command - namely it needs to be updated whenever a new videocore command is added.

andrum99, 2019-01-15:
See the elinux wiki for a starting point for some docs: https://elinux.org/RPI_vcgencmd_usage

andrum99, 2019-01-15:
See https://github.com/raspberrypi/userland/pull/527

JamesH65, 2019-01-16:
Thanks. First draft at docs (actually including vcbdg) on https://github.com/raspberrypi/documentation/tree/JamesH65-patch-vcgencmd-vcdbg-docs

andrum99, 2019-01-17:
Interesting - hardware support for animated GIFs. I suppose that's to be expected given the target market at the time the VC4 was developed.

Some review comments on the vcgencmd page in development:

`Displays the supported and deected state of the official camera`

Doesn't explain what 'supported' means. Suggest:

`Displays whether the current firmware supports the camera module, and whether the VideoCore has detected that a camera module is currently attached.`

(Wording not perfect but you can probably leave that to a copy-editor).

Also, for get_throttled - replace 'under-voltage' with 'under-voltage detected'.

For 'get_lcd_info' it would be worth noting which displays this refers to - HDMI is one of them as my current Pi is returning '1920 1080 24' and only has an HDMI-connected display attached.

JamesH65, 2019-01-21:
Thanks, updated.

andrum99, 2019-02-22:
I'm not sure adding the documentation for the vc... commands under Raspbian is right. These utilities can be used on other distros. I would expect the Raspbian section of the docs to be used for Raspbian-specific stuff. The best fit out of the existing documentation categories is probably "Hardware".

I've made my changes to the vcgencmd binary dependent on this documentation landing, since my changes to the binary don't actually do anything particularly useful on their own. I've added the proposed URL for the vcgencmd documentation to the usage.

JamesH65, 2019-02-25:
There's a problem here that the categories we currently have don't really reflect what I'd like to add in some cases.  Really need to assess that but because of the first page being different to the rest of the repo (pretty graphics), its a bit awkward to change. Will look in to it.

andrum99, 2019-02-25:
Perhaps that's an issue to add to the wider review of the website which seems to be underway.

andrum99, 2019-02-25:
I've thought about this a bit, and I think the best way forward is probably just to add the docs for vcgencmd et al to the existing 'Raspbian' category, then leave any reorganisation of the docs to a future task. (I'm also not entirely sure how much of the various utilities the other distributions are using).

bennuttall, 2019-02-25:
Agreed

lurch, 2019-12-09:
Closed by #1077

closed 2019-12-09 by lurch