raspberrypi/documentation #1078: access-point.md  patch

malessandrini, 2019-02-07:
Hi,
in "configuration/wireless/access-point.md", when setting dhcpcd.conf for static ip address on wlan0, it does not work for me as is, it seems it needs at least the following:
```
interface wlan0
static ip_address=192.168.4.1/24
static routers=192.168.4.1
static domain_name_servers=192.168.4.1
nohook wpa_supplicant
```

Thanks
Michele Alessandrini

JamesH65, 2019-02-07:
I suspect that whole document need revamping for the latest Raspbian. I think there might be some PR's waiting to go through on this page, but will leave this open as it is something that needs looking at.  Thanks for the report.

XECDesign, 2019-03-11:
Could you please provide a bit more information? In what way does it not work?

After hostapd starts, what's the output of
```
ip addr show wlan0
ip route
sudo journalctl -u dhcpcd
```

malessandrini, 2019-03-11:
It did not work because wlan0 was not assigned the given IP address, so all the remaining tools depending on it did not work. Currently I don't have a Raspberry 3 any more, so I can't connect to it via ethernet leaving wlan0 unconfigured. Anyway it was a fresh installation or Raspbian Strecth Lite.

XECDesign, 2019-03-11:
While updating the docs, I tried the example as is and the IP was set as expected. Without exact output, it's hard to diagnose what went wrong in your case.

malessandrini, 2019-03-11:
I'm sorry, that's strange, I'll try to test it again as soon as possible. Thanks for now.

XECDesign, 2019-03-11:
If you do, please use this version https://github.com/XECDesign/documentation/blob/wifi-ap-update/configuration/wireless/access-point.md (sent as a PR)

JamesH65, 2020-11-19:
I believe the changes to the page over the last year  will have taken this in to account.

closed 2020-11-19 by JamesH65