raspberrypi/documentation #1079: FAQ update request

robthez, 2019-02-07:
After having a discussion with a new Pi user, I feel we need to update some of the wording on the FAQ, specifically the section on SD cards:

"What size of SD card can it support?

We have tried cards up to 128GB, and most cards seem to work OK. You can also attach a USB stick or USB hard drive to provide extra storage."

As the result of this (emotionally draining) conversation, I think it needs to be clarified that a 128GB card is not necessary for the vast majority of Raspberry Pi users. Even if it spells it out in the previous point.

JamesH65, 2019-02-07:
OK, will do that now. I'll PR it and post you as a reviewer.

lurch, 2019-06-08:
Fixed by #1080

closed 2019-06-08 by lurch