raspberrypi/documentation #1081: Camera connector pinout is reversed in CM3 IO board

ebenolson, 2019-02-07:
In this image, you can see that on the camera adapter board, the differential pairs are on the left side of the connector (facing into the connector). However, on the IO board, the differential pairs have swapped to the right side.

This is necessary of course for the signals to match up, however it means the numbering of the connector needs to be reversed in the schematic.

[image]

lurch, 2019-02-08:
Is this a duplicate of #847 ?

ebenolson, 2019-02-08:
Looks like it! Same thing happened to me, guess I'll close this and add my
vote there.

ebenolson, 2019-02-08:
Actually, reopening as it's not really a duplicate - same mistake, but affects a different board and schematic file.

JamesH65, 2020-11-19:
@jimbojr Was this schematics issue fixed?

closed 2021-06-17 by aallan