raspberrypi/documentation #1083: Formatting of bash script on building.md

willdean, 2019-02-08:
There's a problem with the way the bash snippets are laid-out on https://www.raspberrypi.org/documentation/linux/kernel/building.md in that some of the commands get wrapped, and it's impossible to tell which things are commands and which are just bits of long lines.

See this, for example, which is three commands but (very confusingly, if you're just trying to carefully follow instructions) looks as though it's four, with a space between the last hyphen and the last word.

[image]

Clearly making the box wider would help, but perhaps part of the difficulty is the bash prettifier which may be adding prettiness but is subtracting usability?

I'm not sure what the best way to fix this is, hence the issue rather than a PR

lurch, 2019-02-10:
Yup, see also the discussion about this in #417

willdean, 2019-02-11:
Thanks @lurch - depressing to see people have been caught out by this for several years.   I think a single line CSS change would improve things - I will suggest it on that other issue

lurch, 2019-09-19:
Closing this as a duplicate of #417

closed 2019-09-19 by lurch

grega, 2019-11-19:
A fix has been pushed for this:

[image]

JamesH65, 2019-11-19:
Forgive my ignorance, what change am I supposed to be seeing or has this not hit the website yet? All I'm still getting the wrapped text, no scroll bar.  And anyone know why = and - seem to have a difference background colour?

bennuttall, 2019-11-19:
What OS are you on James? Might be a less visible scrollbar

LGTM on Ubuntu 18.04: 

[image]

JamesH65, 2019-11-19:
Windows 10 running Chrome. Probably some caching issue somewhere?
Working on a virtual Ubuntu running Chrome which us running on the same laptop.

JamesH65, 2019-11-19:
Just updated Chrome and it started working.....

JamesH65, 2019-11-19:
Defo not right for some reason - no scroll bars. Shut down Chrome, restarted. I wonder if I need to reboot. Will try that and get back.

lurch, 2019-11-19:
Yeah, I suspect it's a browser-caching problem. I had to force-refresh in order to get the "newness".

> And anyone know why = and - seem to have a different background colour?

Hmmm, it's a very subtle difference, which I didn't notice at first. But now that I've seen it I can't un-see it :confused: 
There's also the different background-colour on slash characters as well (as seen in Ben's screenshot). Any ideas @grega ?

JamesH65, 2019-11-19:
shift-F5 seems to have done the trick - AIUI that's a forced refresh. Weird that shutting Chrome down and restarting didn't fix it.

lurch, 2019-11-19:
...hmmm, looking at the source it seems that there's e.g. `<span class="token operator">/</span>` wrapped around the `-`, `=` and `/` characters, so I guess for some reason they must have a different background colour set in CSS? I dunno much about CSS and styling...

bennuttall, 2019-11-19:
Syntax highlighting is done by a third-party library, IIRC prism: https://prismjs.com/

Might be possible to customise it, these are probably just the default settings.

lurch, 2019-11-19:
Or it could just be part of the https://prismjs.com/known-failures.html

grega, 2019-11-19:
@JamesH65 definitely a browser cache issue, glad it's all resolved now (I purged the edge CDN cache when I published the change so now it's just down to browsers to purge their own caches).

For the highlighting, yep as Ben said it's a Prism thing. There are some general improvements planned for the documentation pages in the near future so I'll pass these sorts of tweaks / improvements on to that team.