raspberrypi/documentation #1087: [ssh] section is not included in jail.conf

jftuga, 2019-02-17:
https://www.raspberrypi.org/documentation/configuration/security.md

This page states:

_Look for the section on [ssh]. It will look something like this._

    [ssh]
    enabled  = true
    port     = ssh
    filter   = sshd
    logpath  = /var/log/auth.log
    maxretry = 6

However, this section is not actually located within the `/etc/fail2ban/jail.conf` file.

Also, the `/etc/fail2ban/filters.d/sshd.conf` file does not exist, but is mentioned in the next paragraph.

This occurs on a fresh install (with all updates applied).  The base image I used was 2018-11-13 Raspbian Stretch Lite.

lurch, 2019-02-17:
fail2ban isn't installed in Raspbian by default, so maybe the config-files provided by the package have changed since that article was first written? :man_shrugging:

jftuga, 2019-02-18:
Instead of:

    Look for the section on [ssh]. It will look something like this.

What about having something like this instead?

    If not already there, add the following [ssh] section to your jail.local file.

JamesH65, 2019-02-18:
Yes, looks like this has changed since I wrote that page. Can people confirm that adding that ssh section actually does what is intended? If so will make the changes or take a PR with the above change.

jftuga, 2019-02-18:
I am testing with this in my `jail.local` file:

    [ssh]
    enabled  = true
    port     = ssh
    filter   = sshd
    logpath  = /var/log/auth.log
    maxretry = 3
    bantime = 120

I ssh'd to the system and typed in an incorrect password 3 times.  I was then disconnected.  When I tried to ssh in again, I got this message:

    ssh: connect to host w.x.y.z port 22: Connection refused

After waiting `120` seconds, I was able to ssh to the system and connect with the correct password.

JamesH65, 2020-11-19:
Updated on master.

closed 2020-11-19 by JamesH65