raspberrypi/documentation #1089: Consider moving more forum stickys into the documentation

andrum99, 2019-02-22:
The forums have a number of sticky posts with various amounts of documentation in them. My proposal is to move the first post in the sticky into the documentation repo. This will allow easier updates. In the process, any changes to the procedures and advice in the first post in the sticky can be collected up into a single page. I am not proposing to remove the stickys from the forums - these should stay as users can add questions and problems to the end of them as at present.

JamesH65, 2019-02-25:
Which stickies do you think need to be moved in to the docs? I#m guessing troubleshooting for one?

andrum99, 2019-02-25:
Any which people feel appropriate. There is one that already points to a page in the docs. The rest have various how-to's that advise folk what to do in certain situations - they're not all in the Troubleshooting forum. I'm thinking of the ones in the General Discussion and Troubleshooting forums since those are the ones I have looked at, but there are likely to be others elsewhere.

I think this may be one for the wider website review as well.

JamesH65, 2019-02-25:
What website review? Nothing on radar here. But it might be my bad radar.

andrum99, 2019-02-25:
They're going to be doing a survey of folk about the website. They've started off collecting expressions of interest along with some basic demographic data. I don't know exactly what the intent is, but they are looking at using this to make some sort of improvements to the website. I can't remember where I saw the link to register (it's wasn't a blog post), but I have registered my interest.

bennuttall, 2019-02-25:
Who are they?

andrum99, 2019-02-25:
RPF/T - here is the twitter post:

https://twitter.com/Raspberry_Pi/status/1098956488599658497

JamesH65, 2019-02-25:
News to me....methinks I might need to talk to someone.

JamesH65, 2019-02-25:
Apparently its the webteam asking the questions, not the comms team, but no-one has mentioned it to me. So who knows what is going on. Will try and find out.

JamesH65, 2019-02-26:
Just remembered, we had an internal email conversation a month ago about adding a new troubleshooting section in the docs, which would be a good place to move a lot of the stickies since they are mostly in that area. I've ressurected it so will hopefully be able to report progress in the near future.

JamesH65, 2019-02-28:
I've started on a troubleshooting section. Very much in draft protype not at all ready stage.

https://github.com/raspberrypi/documentation/blob/JamesH65-patch-troubleshooting/troubleshooting/readme.md

lurch, 2019-11-06:
Closing, as it seems the person asking this has since deleted their GitHub account.

closed 2019-11-06 by lurch