raspberrypi/documentation #1093: RPi Update broke Access Point setup

GizmoEV, 2019-03-06:
I don't know if this is related to https://github.com/raspberrypi/documentation/issues/1078

I have an RPi 3B+ running an access point as described in the first part of https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md except that the uplink is a WiFi USB dongle rather than eth0. The RPi links a WeatherFlow Smart Weather station hub to the internet. Also ArchiveSW, a weather database with web interface, runs on the device. After the update I could no longer reach the web page of ArchiveSW, in addition to the access point interface not being associated with anything.

When I did a new install and went through the steps to setup the access point at the step when I ran "sudo systemctl start hostapd" it gave some error about hostapd.???? being masked or some such thing. Any suggestions on how I can get it working again, short of doing a fresh install without updating?

Thank you,
David

JamesH65, 2019-03-06:
@XECDesign Is this related to the recent hostapd changes?

lurch, 2019-03-06:
> the recent hostapd changes

For context: https://www.raspberrypi.org/forums/viewtopic.php?p=1437905#p1437905

XECDesign, 2019-03-06:
I haven't used hostapd in years, so I won't be of much help here.

Others seems to report that the update has solved their problems and works well.

The exact error message may be useful.

GizmoEV, 2019-03-07:
Thank you, @lurch  I followed the steps in https://www.raspberrypi.org/forums/viewtopic.php?p=1437905#p1438251 and that fixed things. The only thing that hasn't worked on my test RPi 3B+ is the reboot after "sudo rpi-update" which has been running for over an hour and hasn't booted to the desktop. Drive activity light is on almost constant so I won't be running rpi-update on my production unit for a while.

JamesH65, 2019-03-07:
Just a note, that you should never run rpi-update on a production device - its beta quality sofware, and there is no guarantee it will work.

XECDesign, 2019-03-07:
I think this snippet of postinst is the cause here:
```
if [ -x /bin/systemctl ] && [ "$1" = configure ]
then
    DAEMON_CONF=
    . /etc/default/hostapd
    if [ -z "$DAEMON_CONF" ] && ! systemctl --quiet is-active hostapd.service
    then
	    systemctl mask hostapd.service
    fi
fi
```

What happens if you run this?
```
sudo systemctl unmask hostapd.service
sudo systemctl start hostapd.service
```

ingegno, 2019-03-10:
@XECDesign that fixed for me the hostapd.service is masked error. 
I was following [wireless AP doc](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md), section Start it Up. Update that manual?

XECDesign, 2019-03-10:
@ingegno It doesn't seem to fix it for everyone.

I'm in the process of running through the instructions myself and seeing what other problems there may be.

GizmoEV, 2019-03-14:
I've been trying several things to get this to work and started over with a new setup of NOOBS minimal install with desktop. I did not let the GUI do the update and instead followed the directions in https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md. After the update and upgrade steps everything went find until the 'sudo systemctl start hostapd' step.

```
pi@raspberrypi:~ $ sudo systemctl start hostapd
Failed to start hostapd.service: Unit hostapd.service is masked.
pi@raspberrypi:~ $ sudo systemctl status hostapd
● hostapd.service
   Loaded: masked (/dev/null; bad)
   Active: inactive (dead)
pi@raspberrypi:~ $ sudo systemctl unmask hostapd
Removed /etc/systemd/system/hostapd.service.

```

After the unmask line the rest of the setup went fine except that after reboot hostapd doesn't start automatically. Here is what I get after a reboot.

```
pi@raspberrypi:~ $ sudo systemctl status hostapd
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
pi@raspberrypi:~ $ sudo systemctl start hostapd
pi@raspberrypi:~ $ sudo systemctl status hostapd
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-03-13 20:24:27 PDT; 15s ago
  Process: 1042 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=0/SU
 Main PID: 1044 (hostapd)
   CGroup: /system.slice/hostapd.service
           └─1044 /usr/sbin/hostapd -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf

Mar 13 20:24:27 raspberrypi hostapd[1042]: Using interface wlan1 with hwaddr b8:27:eb:12:08:f1 and ssid "ArchiveSW"
Mar 13 20:24:27 raspberrypi hostapd[1042]: random: Only 16/20 bytes of strong random data available from /dev/random
Mar 13 20:24:27 raspberrypi hostapd[1042]: random: Not enough entropy pool available for secure operations
Mar 13 20:24:27 raspberrypi hostapd[1042]: WPA: Not enough entropy in random pool for secure operations - update keys la
Mar 13 20:24:27 raspberrypi hostapd[1042]: wlan1: interface state UNINITIALIZED->ENABLED
Mar 13 20:24:27 raspberrypi hostapd[1042]: wlan1: AP-ENABLED
Mar 13 20:24:27 raspberrypi systemd[1]: Started Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Mar 13 20:24:39 raspberrypi hostapd[1044]: wlan1: STA b0:38:29:7b:fd:9b IEEE 802.11: associated
Mar 13 20:24:39 raspberrypi hostapd[1044]: wlan1: STA b0:38:29:7b:fd:9b RADIUS: starting accounting session C55256964F50
Mar 13 20:24:39 raspberrypi hostapd[1044]: wlan1: STA b0:38:29:7b:fd:9b WPA: pairwise key handshake completed (RSN)
pi@raspberrypi:~ $
```
Any suggestions on what to do to get hostapd to start automatically?

[Note: I have a USB WiFi for the uplink. Raspbian insists on this being wlan0 so wlan1 is the RPI 3B+ onboard WiFi.]

ingegno, 2019-03-14:
@GizmoEV  To have hostapd start at startup I did

    sudo update-rc.d hostapd defaults
    sudo update-rc.d hostapd enable

You do need to give it some time at startup before it works.

XECDesign, 2019-03-14:
> Any suggestions on what to do to get hostapd to start automatically?

That ingegno said should do it, but the documentation PR I sent has this line:
```
sudo systemctl enable hostapd
```

GizmoEV, 2019-03-16:
Thank you, @ingegno and @XECDesign 
That fixed the issue. When going through the directions at https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md I added in your suggestion(s) just before reloading hostapd and things worked. Hopefully someone more knowledgeable than me will update the access-point.md directions. FWIW, I'm running on an RPi 3B+.

lurch, 2019-03-16:
@GizmoEV Something like #1097 ? :wink:

closed 2019-04-30 by JamesH65

qliang573, 2019-05-20:
Just a reminder for Raspberry Pi 3B+ users who is following this instruction: https://elinux.org/Talk:RPI-Wireless-Hotspot.
The 802.11ac seems like not supported, and the channel 6 will not work. 

Thanks for @lurch 's suggestion to update the Linux kernel, my issue is solved! Thanks to everyone in this post as well!

PR #1097 (2019-04-30): Update WLAN AP documentation
Fixes #1093, fixes #1095