raspberrypi/documentation #1095: Access Point Issue

MrMatttw, 2019-03-07:
I have an RPi 3B+ running the Debian lite system, downloaded from the official website, and configuring the access point as described in the https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md. However, when I configured the file `/etc/hostapd/hostapd.conf` and `/etc/default/hostapd`, the service cannot start. Here are the information:
```
Failed to start hostapd.service: Unit hostapd.service is masked.
```
Then, I used the command `sudo hostapd /etc/hostapd/hostapd.conf`, and the information is:
```
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr xx:xx:xx:xx:xx:xx and ssid "RaspberryPi"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
```
All the other configuration are the same as the documents in your tutorial.
Here are my ifconfig output, and I can get internet access via eth0:
```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet xxx.xxx.xxx.xxx  netmask xxx.xxx.xxx.xxx  broadcast xxx.xxx.xxx.xxx
        inet6 xxxx::xxxx:xxxx:xxxx:xxxx  prefixlen 64  scopeid 0x20<link>
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 9533  bytes 613526 (599.1 KiB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 435  bytes 54436 (53.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 357  bytes 42306 (41.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 357  bytes 42306 (41.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 30  bytes 4694 (4.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
Could you please tell me how can I solve this problem? Thank you very much

XECDesign, 2019-03-07:
Duplicate of https://github.com/raspberrypi/documentation/issues/1093 ?

MrMatttw, 2019-03-07:
Maybe not the same issue

MrMatttw, 2019-03-07:
> Duplicate of #1093 ?

Maybe not the same issue

MrMatttw, 2019-03-07:
> Duplicate of #1093 ?

After I execute the solution for #1093, the problem is still the same.

XECDesign, 2019-03-07:
What's the content of  `/etc/default/hostapd`?

MrMatttw, 2019-03-08:
> What's the content of `/etc/default/hostapd`?
> 
> EDIT: Actually, I don't think that matters. I'll have to run through the instructions and see what happens.
The /etc/default/hostapd file is shown below:
```
# Defaults for hostapd initscript
#
# See /usr/share/doc/hostapd/README.Debian for information about alternative
# methods of managing hostapd.
#
# Uncomment and set DAEMON_CONF to the absolute path of a hostapd configuration
# file and hostapd will be started during system boot. An example configuration
# file can be found at /usr/share/doc/hostapd/examples/hostapd.conf.gz
#
DAEMON_CONF="/etc/hostapd/hostapd.conf"

# Additional daemon options to be appended to hostapd command:-
#       -d   show more debug messages (-dd for even more)
#       -K   include key data in debug messages
#       -t   include timestamps in some debug messages
#
# Note that -B (daemon mode) and -P (pidfile) options are automatically
# configured by the init.d script and must not be added to DAEMON_OPTS.
#
#DAEMON_OPTS=""
```

akdigitalcss, 2019-03-10:
Exact same issue for me

> ```
> # Defaults for hostapd initscript
> #
> # See /usr/share/doc/hostapd/README.Debian for information about alternative
> # methods of managing hostapd.
> #
> # Uncomment and set DAEMON_CONF to the absolute [...]

Did you ever get this resolved I am having the exact same issue. I have spent the past 3 days even using some of the older lite builds and nothing but issues. 

> Failed to start hostapd.service: Unit hostapd.service is masked.

Is where I am getting stuck at, even after unmasking the service it still does not fix the issue so instead I am trying to focus on WHY its failing to start in the first place.

XECDesign, 2019-03-10:
Unmasking the service worked for me.

What's the output of the following commands?
```
sudo systemctl status hostapd
sudo systemctl unmask hostapd
sudo systemctl start hostapd
sudo systemctl status hostapd
sudo journalctl -u hostapd
```

PR #1097 (2019-04-30): Update WLAN AP documentation
Fixes #1093, fixes #1095
closed 2019-04-30 by JamesH65