raspberrypi/documentation #1096: Access Point Issue when creating a WiFi Router with OpenVPN.

akdigitalcss, 2019-03-10:
For those having an issue creating a router for the sole purpose of connecting to it to use a VPN I have the temporary fix to all of your problems. I have seen many people trying to do this and absolutely no valuable information being passed back from the authorities within this repo.

I have spent the past 3 days solid trying to fix this and I have just found the solution;

Use this image - http://director.downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/2017-07-05-raspbian-jessie-lite.zip

This guide to setup your access point - https://pimylifeup.com/raspberry-pi-wireless-access-point/

This guide to setup openvpn - https://pimylifeup.com/raspberry-pi-vpn-access-point/

Never will I be updated to the latest version of Raspian again. On a side note, don't be looking for guides specifically tailored to your VPN provider just download their .ovpn files and follow the guide normally.

Me personally I set the `#autostart="all"` line to `autostart="all"` and drop whatever .ovpn files I want to connect to, rename them to .conf and use the same `auth-user-pass` line in each of them.

IF YOU DO NOT rename your config files from .ovpn to .conf they will not automatically connect.

JamesH65, 2019-03-10:
The documentation has not been able to keep up with the changes to the latest versions of Debian/Raspbian. I just haven't had time to find the correct process and document it. If anyone else has got it all working with the latest Raspbian, then plaese submit a PR with the changes.

lurch, 2019-03-10:
On a side note, recommending people to use an image that's almost 2 years old is never a good idea :confused:  Especially as I assume you want it to be secure (since you're using a VPN).

akdigitalcss, 2019-03-10:
> On a side note, recommending people to use an image that's almost 2 years old is never a good idea 😕 Especially as I assume you want it to be secure (since you're using a VPN).

With SSH disabled or the usual basic security measures in place (fail2ban, strong password or key, different port that was closed via its route to the internet) what exactly could be done?

JamesH65, 2019-03-11:
Well, a two year old Raspbian will have security holes that have been fixed in subsequent releases. Those holes may be a problem even if you take all the normal security precautions.

akdigitalcss, 2019-03-11:
> Well, a two year old Raspbian will have security holes that have been fixed in subsequent releases. Those holes may be a problem even if you take all the normal security precautions.

Right but if my Raspberry Pi is not reachable on the WWW then that can only really leave me vulnerable to a local attack? As in if somebody was to sit within X distance of my hotspot, break in and then take advantage of said holes to possibly gain further access?

JamesH65, 2019-03-11:
No idea, I am not a security expert, and not a hacker!

akdigitalcss, 2019-03-11:
> No idea, I am not a security expert, and not a hacker!

Right well when I did the obvious and unmasked the hostapd service I was still running into an issue when trying to connect to the AP, maybe it was user error or maybe I missed something off after retrying so many times. If I get chance this week I will try it yet again and if successful submit the additions that need to be made.

closed 2021-06-17 by aallan