raspberrypi/documentation #1098: mount.cifs gives erroneous "host is down"

roykimbrell, 2019-03-11:
"mount.cifs //IP/share /mnt/point -o user=uname" results in an erroneous "host is down" error notice. The solution, as noted at several locations (https://askubuntu.com/questions/915549/16-04-cifs-host-is-down-but-they-are-not, https://www.raspberrypi.org/forums/viewtopic.php?t=210311, and https://access.redhat.com/discussions/3002961), is to add the option vers=1.0. As in

mount.cifs //IP/share /mnt/point -o user=uname,vers=1.0

lurch, 2019-03-11:
One of the comments on the askubuntu page you linked to https://askubuntu.com/a/1070656 suggests that the SMB version used depends upon which server-host you're connecting to, so presumably adding `vers=1.0` isn't a universal solution?
But it definitely sounds like the sort of thing that would be worth adding to https://github.com/raspberrypi/documentation/blob/master/remote-access/samba.md

roykimbrell, 2019-03-12:
Yeah. That's the place.

andrum99, 2019-05-08:
Reading https://askubuntu.com/questions/915549/16-04-cifs-host-is-down-but-they-are-not reveals that this error results from 2 things: SMB protocol version mismatch, and the cifs client on Linux returning a misleading error message. The solution is not as simple as adding vers=1.0, since it depends which version of the SMB protocol the server supports. The need to add vers=x.0 to the mount should certainly be documented, with an explanation that the user may need to try different values for x.

JamesH65, 2020-11-19:
Fixed by #1739

closed 2021-06-08 by JamesH65