raspberrypi/documentation #1101: Description of switching to a different username is not complete

KimSJ, 2019-03-16:
Your description of how to create a usable username other than pi (in  documentation/configuration/security.md) is not complete. The new user will have a very limited set of groups! To rectify that, the following bash command works well:

```
for i in `grep -E "(:|,)pi(:|,|$)" /etc/group|cut -f1 -d:` ; do
	sudo addgroup <new_username> $i
done
```
Obviously this should be run before you delete the pi user!

JamesH65, 2019-03-16:
Interesting idea. WIll talk about in Monday in the office.

lurch, 2019-03-18:
Related to #1092 ?

lurch, 2019-10-02:
@KimSJ #1092 has now been merged - does it adequately address your concerns?

JamesH65, 2020-11-19:
I believe this has been fixed by the merge mentioned,

closed 2020-11-19 by JamesH65