raspberrypi/documentation #1107: /etc/sudoers  direct file editing  instructions need updating/correcting.

Nightblade, 2019-03-22:
Various locations in the documentation suggest editing `/etc/sudoers` directly, with `nano` or `vi`.  This is highly inadvisable as it carries the worst-case risk of rendering the system unusable.  The preferred method is to use `visudo` which automatically checks for errors, and to only perform changes in the `/etc/sudoers.d/` directory, instead of the main file.

I noticed "dubious" instructions these two pages, but there may be more:
https://github.com/raspberrypi/documentation/blob/master/linux/usage/users.md
https://github.com/raspberrypi/documentation/blob/master/configuration/security.md

jmpcm, 2019-11-10:
Also, there is an extra space between "NO PASSWD" on the line to add to 010_bob-nopasswd. It should be:

`echo 'bob ALL=(ALL) NOPASSWD: ALL' | sudo tee /etc/sudoers.d/010_bob-nopasswd`

The extra space renders using sudo unusable, because there is an error parsing the file. The only way, in my case, to solve this was to mount the CM module (or SD card) and edit the file `010_bob-nopasswd`, on another Linux machine.

lurch, 2019-11-10:
@Nightblade 
> Various locations in the documentation suggest editing /etc/sudoers directly, with nano or vi.

AFAICT, it seems https://github.com/raspberrypi/documentation/blob/master/linux/usage/root.md is now the only page remaining that suggests that?
Please feel free to submit an appropriate Pull Request.

@jmpcm 
> Also, there is an extra space between "NO PASSWD" on the line to add to 010_bob-nopasswd.

Was there an error in #1269 ? :confused:

jmpcm, 2019-11-11:
Indeed, the error was introduced in #1269.

JamesH65, 2019-11-11:
What specifically is the fix needed - I'll do it once I know. ie where is the bad extra space?

JamesH65, 2019-11-11:
#1331 This OK?

jmpcm, 2019-11-12:
Yes, that will fix it :) Thank you!

StoneTrue32, 2020-05-07:
This has not been fixed at https://www.raspberrypi.org/documentation/configuration/security.md as of 5/7/2020.  Instead of "sudo nano /etc/sudoers.d/010_pi-nopasswd" instead use "sudo visudo... ".

(first time using Github, please be kind!)

solt87, 2020-05-18:
As @StoneTrue32 mentioned, https://www.raspberrypi.org/documentation/configuration/security.md still contains `sudo nano...`.

While we're at it, it's worth mentioning that editing files (other than sudoers-files) as root generally should happen using something like `sudoedit /path/to/file` or `EDITOR=/path/to/editor sudoedit /path/to/file`.

JamesH65, 2020-05-18:
Please send in a PR with the appropriate changes as I doubt I will get it right myself. I am not an expert on this.

JamesH65, 2020-05-18:
Just been pondering this as never even heard of sudoedit. There are quite a few references to "sudo nano" in the docs, should they ALL be replaced with sudoedit?

aallan, 2020-05-18:
Using sudoedit is essentially telling the shell to use whatever editor is defined by the EDITOR environmental variable to edit the file. There are ways to lock down sudo'ers rights so that they can only edit rather than just generically sudo to root, but that seems a bit beyond what we're trying to do with the Raspberry Pi documentation. I'm not sure it's worth complicating things by introducing sudoedit into the docs.

StoneTrue32, 2020-05-18:
The issue is using any editor besides visudo is risky.  "visudo" parses the changes and ensures you do not make a syntax error that goofs up sudo rights, which can be painful to fix, especially on a headless unit.  See [https://unix.stackexchange.com/questions/27594/why-do-we-need-to-use-visudo-instead-of-directly-modifying-the-sudoers-file](url)  for details.  I am mad at myself because I KNOW this, and still, in a hurry following the page, blindly used "nano" and had to reimage my SD card.

andrum99, 2020-05-18:
I've filed #1550 to fix this issue in all files in this repo.

My analysis is below:

Sudoedit is not required here. From what I can gather, the correct fix is to use `sudo visudo ...` in place of `sudo nano ...` when editing any of the sudoers files, as @StoneTrue32 says.

I've just had a look on Raspbian, and Debian with Pi Desktop in a VM. Despite its name, visudo does not (always) run vi. Per the man page, it uses /usr/bin/editor, which on Debian and Raspbian is linked to /etc/alternatives/editor. On both systems this seems to default to nano, although I may already have run something else that set that default. So I'm satisfied this is not going to be a problem for relatively inexperienced users. (As you can probably tell, I don't like vi, but let's not go down that particular rabbit hole).

lurch, 2020-05-22:
Fixed in #1550 - thanks for everyone's contributions.

closed 2020-05-22 by lurch