raspberrypi/documentation #1114: How to create local hotspot on raspberry pi 3b to share files on android

ErShakirAnsari, 2019-03-28:
I want create a local file shareable service using raspberry, below are the requirements:

- **Create wifi/hotspot on raspberry (without any internet)**
- **Connect my android phone to created wifi**
- **Share files between raspberry and my phone.**

> ### Think any file sharable app like xender or share it, their working principle is same as I want.

**Below are links which I've tried so far:**
- https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md
- https://elinux.org/RPI-Wireless-Hotspot

I was able to create a wifi and my android device is showing it but not getting connected as shown is below screenshot.

[image]

lurch, 2019-03-28:
I think this sounds like a question for the [forums](https://www.raspberrypi.org/forums), rather than the documentation repo?

JamesH65, 2019-03-28:
Indeed, please ask your question on the forums as linked.

closed 2019-03-28 by JamesH65