raspberrypi/documentation #1119: Noobs boot on 256 and 512 GB sc-cards

ThomasFreedman, 2019-04-10:
NOOBS 3.0.0 - Used gparted 0.25,9 on Debian Linux (Stretch v 9.8, OS kernel v 4.14) to format the entire card with fat32. Copied all of the unzipped files downloaded from the Raspberry Pi website (https://www.raspberrypi.org/downloads/noobs/) to the root of the sd card. Booted in RPi3 (not +), but got an error on 256 stating it could not resize the fat32 partition. I answered that dialog and another progress bar appeared and never got beyond that.

On the 512GB sd card I just got a kernel panic and system just froze.

This behavior  was repeatable. Both sd cards were Samsung EVO. The 512GB card was green & white, class 3, write speed 90MB/s. The 256GB sd card is red and white, but had same specs.

These cards don't appear to be usable on the RPi3, at least not using NOOBS to install the OS image.

lurch, 2019-04-10:
@ThomasFreedman You should report this on the NOOBS repo https://github.com/raspberrypi/noobs/ not on the documentation repo :wink:

ThomasFreedman, 2019-04-10:
You're right, but the link to this repo was readily at hand and it was late and didn't think about where I was posting it. I'll copy it over there. This issue can be closed here. 

I would say tho that this sdxc documentation should be easy to find and appear on even the simple / quickstart guides tho.

lurch, 2019-04-10:
> I would say tho that this sdxc documentation should be easy to find and appear on even the simple / quickstart guides tho.

That's possibly a valid point... when I first wrote that [SDXC page](https://github.com/raspberrypi/documentation/blob/master/installation/sdxc_formatting.md), SDXC cards were a lot less common than they are now! :slightly_smiling_face:  
ping @JamesH65 and @XECDesign

lurch, 2019-11-06:
Closing, as the relevant question has migrated to the NOOBS repo.

closed 2019-11-06 by lurch