raspberrypi/documentation #1121: Missing Network Documentation

pkropf, 2019-04-16:
While there is great documentation on how to configure and use a Raspberry Pi on a Wifi network, there doesn't seem to be any for how to configure for an ethernet based network. Can this be remedied?

XECDesign, 2019-04-16:
What sort of information are you after? Just how to set up static ip or something more? On a typical network, everything will just work out of the box, so there's not much to document.

pkropf, 2019-04-16:
There's lots of details on how wifi is configured and how to make changes as needed. With this documentation, a user can begin to debug any issues that they may have. 

With ethernet, there is nothing documented on the . While the default configuration may work out of the box for many people, it doesn't always. Without having any details on how to examine the configuration, what the configuration does and where to make changes as needed, a user is left with asking google. The information that provided there is just wrong. And since the master location for configuration (https://www.raspberrypi.org/documentation/configuration/) doesn't provide any help, it leaves users frustrated and confused.

I spent 3 hours today trying to determine why there was no search entry in the /etc/resolv.conf. Much of that time was trying to find where the ethernet adapter is configured on the Raspberry Pi.

pkropf, 2019-04-17:
The particulars of my problem were that the local /etc/resolv.conf does not contain a search parameter even though the local DHCP server provides it. Since this is working for other systems, Linux and Windows based, my first step is check on the network configuration on the Raspberry pi. Looking in /etc/networks/interfaces didn't show any configuration information. Then looking for netplan configuration, which wasn't there. Instead of trying to guess where else to look, I went to the Raspberry Pi configuration documentation page to read up on how ethernet networking is configured. But there's nothing there to help.

There's all sorts of great documentation there to help people learn and deal with any issues that they may encounter. Ethernet networking is missing.

andrum99, 2019-04-25:
Did you try the Debian wired networking docs? Raspbian is based on Debian, so it should work the same for both. Although perhaps the desktop GUI apps in Raspbian are set up to do things differently from stock Debian.

XECDesign, 2019-04-26:
The main difference is that Debian recommends NetworkManager, whereas we use dhcpcd.

andrum99, 2019-04-26:
So at the bar minimum we should probably have a page that says we use dhcpcd. What are the other differences? I will write a page documenting the differences between us and stock Debian and file a PR.

JamesH65, 2019-04-26:
Thanks, probably worth having a page for ethernet, but most stuff I pressume will just be links to the Debian networking, although we are trying to reduce the amount of offsite links.

andrum99, 2019-04-26:
I don't know what shape the Debian docs are in - some of the stuff I looked at was out of date. I won't be linking to stuff that doesn't apply to the Pi - I will check it out first. I may end up just documenting most of it, depending on how much of it there is.

JamesH65, 2019-04-30:
@andrum99 Are ny of the recent PR/changes to do with this? If so can you link them here?

andrum99, 2019-04-30:
Yes - #1128 is the beginnings of documenting the wired side of things

helenlynn, 2019-10-31:
Removed the `ready for copy-edit` label, since the actual content is in https://github.com/raspberrypi/documentation/pull/1128 and that PR itself has this label.

@JamesH65 can we close this, since presumably further work/discussion can take place in #1128?

JamesH65, 2019-10-31:
Usually use "fixes #" in the PR to automagically close the indicated issue when the PR is merged. I'll add it.

helenlynn, 2019-10-31:
Aha. Thanks

lurch, 2019-10-31:
Fixed by #1128

closed 2019-10-31 by lurch