raspberrypi/documentation #1129: wifi headless -> new wpa_supplicant.conf

Travelbacon, 2019-04-27:
Dear all,

The current instructions for headless WiFi configuration are no longer working. There are three more lines needed due the new Debian policty. Also EOL in the wpa_supplicant.conf should be UNIX style. Otherwise the Pi will have no WiFi at boot. I think a small adjustment in the following links should make it working FAQ again.

https://www.raspberrypi.org/documentation/configuration/wireless/headless.md
https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

I would suggest to add something

```
Begin the the configuration file with
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=<country code>*

Replace <country code> with your contry code. Your country code can be found on https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2#Officially_assigned_code_elements

```
And mention that the EOL Conversion must be set to UNIX.

PSOHart, 2019-06-12:
I was going to ask the same thing, the docs should mention to start the file with

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=FR

But I've created the file with Notepad++ on a Windows, and it works without EOL Conversion.

JamesH65, 2019-11-06:
@XECDesign have you any thoughts on this?

XECDesign, 2019-11-06:
Yes, the config on the headless page wouldn't work anymore.

The cli page is not wrong, but if it had the header was included somewhere it would help users who have deleted it by accident.

JamesH65, 2019-11-06:
So do I just need to add 

` ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=FR`

to the top of the file in the docs?

XECDesign, 2019-11-06:
Where 'FR' is replaced with the appropriate ISO3166-alpha2 code.

Yeah, just something that makes it clear that that has to be there. Sometimes people remove it and replace everything with their network block.

Travelbacon, 2019-11-07:
> Yes, the config on the headless page wouldn't work anymore.
> 
> The cli page is not wrong, but if it had the header was included somewhere it would help users who have deleted it by accident.

If you could change it, it would be very welcome.

JamesH65, 2019-11-07:
Just to confirm, you need this at the top of the file, or is the second ctrl_interface now unnecessary?

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=us
ctrl_interface=/var/run/wpa_supplicant
```

JamesH65, 2019-11-07:
OK, have updated the two places mentioned. On master in github, live in an hour or two. Didn't bother with a PR but should be fine. He said, optimistically.

jimlynnjulian, 2020-04-22:
I just tried this on my RPi4 with and without Unix EOL.
Not working.
Also added the ssh file without extension.
Any ideas?

JamesH65, 2020-04-22:
> I just tried this on my RPi4 with and without Unix EOL.
> Not working.
> Also added the ssh file without extension.
> Any ideas?
> 
> begin code
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
> update_config=1
> country=US
> 
> network={
> ssid="Name"
> scan_ssid=1
> psk="Password"
> key_mgmt=WPA2
> }
> end code
> 
> NOTE: just tried with a different card and the Lite version. Still no success.
> I'm replacing an installation that crashed when I tried to auto mount an HDD hub by editing the fstab file.
> The previous installation was made as usual, but with a LAN connection to the router.
> I'll try that again, perhaps with greater success.

You will get a better technical response by posting your question on the forum.

Travelbacon, 2020-05-10:
The documentation is corrected now. Time to close this item.

closed 2020-05-10 by Travelbacon