raspberrypi/documentation #1134: camera.md does not mention flicker avoidance flag added in 2017

george-hawkins, 2019-05-04:
Sorry for not submitting a pull request to cover this.

The [camera.md](https://github.com/raspberrypi/documentation/blob/master/raspbian/applications/camera.md) documentation does not mention the flicker avoidance flag that was merged into `raspivid` in July 2017 (see PR [406](https://github.com/raspberrypi/userland/pull/406)).

This adds the flicker avoidance flag `--flicker` or `-fli` that takes four options `off`, `auto`, `50Hz` and `60Hz` (the default is `off`).

I don't know how one determines whether `50Hz` or `60Hz` is appropriate for one's particular situation but I found setting `-fli auto` dramatically reduced the flicker I saw when using the Raspberry Pi camera module V1 (I haven't tried it with the V2 module) to stream from my Pi to my laptop.

So given this improvement, it definitely seems worth documenting for others. It would also be useful to know if there are any downsides to enabling flicker avoidance (otherwise why isn't `auto` the default rather than `off`). E.g. maybe it has CPU usage consequences or affects things like the maximum FPS.

6by9, 2019-05-16:
The mains frequency varies between countries - see https://en.wikipedia.org/wiki/Mains_electricity_by_country#Table_of_mains_voltages,_frequencies,_and_plugs

Flicker avoidance locks the exposure time to a multiple of the main flicker frequency (8.33ms for 60Hz, or 10ms for 50Hz). This means that you get noisier images as the control algorithm has to increase the gain instead of exposure time should it wish for an intermediate value.
```auto``` can also be confused by external factors, therefore it is preferable to leave it off unless actually required.

JamesH65, 2019-09-18:
Fixed by https://github.com/raspberrypi/documentation/pull/1286

closed 2019-09-19 by lurch