raspberrypi/documentation #1135: TFTP folder update due to package bug(s)

Kasperx, 2019-05-07:
Hi

I would like to correct this doc for usage of the tftp-package du to a known bug. Unfortunately I didn't save the link to that discription, but the solution is pretty easy :)
The tftp-package only reads the folder "/srv/tftp" with access rights for user "nobody" and groud "nogroup", maybe also someone else :)
So one solution might be to create a symlink from "/srv/tftp" to your folder "/tftpboot" (with similar rights) or to use the other folder from beginning on.

Regards
Kasperx

Kasperx, 2019-05-07:
Sry, forgot.
You can see the client (tftp-client) wanting to read files from the folder "/srv/tftp" in the debug output, discriped in your docs. It doesn't care on the edited config-file.

JamesH65, 2019-05-07:
Can you provide a link to the page you believe is incorrect?

Kasperx, 2019-05-10:
Yes -> [docu](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/net_tutorial.md)
[image]

JamesH65, 2019-05-10:
Thanks for that, but I'm a bit confused as to how this could ever have worked given the apparent bug. Do you have a link to the tftp-package bug report?

lurch, 2019-05-10:
Perhaps @Kasperx is using a different TFTP server than the one that `net_tutorial.md` was originally written for? :man_shrugging:

Kasperx, 2019-05-13:
@lurch I installed a tftp server with `sudo apt install tftpd` with verion=0.17-18 on raspbian 9.
@JamesH65 I don't know if the process discriped in docu ever worked  :( In my test it didn't, only with symbolic link or other folder.
Perhapt just the link from website to docu is wrong ?
[Website](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md)

Kasperx, 2019-05-13:
I came to that solution by visiting this [post](https://thwack.solarwinds.com/thread/16628).
Somewhere here...
[image]

And of course the log file with `tail -f /var/log/daemon.lo` (shown in docu) shows nice terminal message with asked folder, where the created folder is NOT the one asked :( No matter what the config file says... (Screenshot coming later)

closed 2021-06-17 by aallan