raspberrypi/documentation #1141: Setting up Access point (AP) and Client interface simultaneously

VinS101, 2019-05-14:
I have completed the official guide on setting up a [Raspberry Pi as an access point in a standalone network (NAT)](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md), and it works perfectly, but I can't join other networks. 

I want my Raspberry Pi to act as an Access point and a client at the same time, and I would like to be able to connect to multiple clients.

Basically, I want the below architecture with my three raspberry Pis:

- Pi 1 joins Pi 2 and Pi 3's networks
- Pi 2 joins Pi 1 and Pi 3's networks
- Pi 3 joins Pi 2 and Pi 1

It would be awesome if the documentation has a section for setting this up. 

Cheers,
Vin

JamesH65, 2019-05-14:
I don't even know if that is possible, and if it is, even less of an idea how to do it. So I won't be writing the docs, but if anyone else want to contribute...

aallan, 2020-05-06:
Theoretically this should work,

https://www.raspberrypi.org/forums/viewtopic.php?p=938306&sid=96aeff9a216668c80d4b0215b036cffc#p938306

but I haven't tested it.

aallan, 2020-05-06:
Although realistically the easiest way to run an AP and client at the same time is to add a second wireless adaptor to your Pi, e.g. https://medium.com/@aallan/adding-a-second-wireless-adaptor-to-a-raspberry-pi-for-network-monitoring-c37d7db7a9bd. Although there I'm not using it as an AP, instead I'm putting it into monitoring mode, the idea is the same.

lurch, 2020-05-06:
Giving @epoch1970 a quick ping to see if he has any ideas about this, given his recent work on our AP documentation :slightly_smiling_face:

epoch1970, 2020-05-06:
I've tried "APSTA" on Raspbian and buildroot, this feature is so constrained it makes no sense and most importantly, it does not work for more than 1 hour (doing nothing). Maybe CYW can make it work, that would be great. Until then, not for me.

VinS101, 2020-05-06:
I was able to make this work by using an additional wireless USB adapter, as @aallan suggested.

aallan, 2020-05-11:
Glad you managed to get that working @VinS101. I'm going to close this issue, but have opened Issue #1537 to get official documentation written up to cover this using the internal wireless card.

closed 2020-05-11 by aallan