raspberrypi/documentation #1142: Can't find schematics for Compute Module

hultqvist, 2019-05-14:
When looking for the Raspberry Pi Compute module schematics I found this page with Raspberry Pi 1,2,3,3B+... as well as Raspberry Pi Zero but no Raspberry Pi Compute Module.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md

It's very easy to draw the conclusion that thee is no official schematics given that page.

I suggest adding a link to the compute module schematics page:

## Raspberry Pi Zero W

- [Revision 1.1](rpi_SCH_ZeroW_1p1_reduced.pdf)

## Raspberry Pi Compute Module

- [Compute Module schematics](https://www.raspberrypi.org/documentation/hardware/computemodule/schematics.md)

```
## Raspberry Pi Compute Module

- [Compute Module schematics](https://www.raspberrypi.org/documentation/hardware/computemodule/schematics.md)
```

lurch, 2019-05-14:
AFAICT the hardware section of the docs seem to be divided up into the "consumer focused" [Raspberry Pi](https://www.raspberrypi.org/documentation/hardware/raspberrypi/) and the "industrial focused" [Compute Module](https://www.raspberrypi.org/documentation/hardware/computemodule/).

@JamesH65 On a side-note, does https://www.raspberrypi.org/documentation/hardware/computemodule/schematics.md need updating to mention the CM3+ ?

JamesH65, 2019-09-18:
Added the requested link.

closed 2019-09-18 by JamesH65

lurch, 2019-09-18:
And I added https://github.com/raspberrypi/documentation/commit/f70d380b6948edd27717248ecc944a2800db0a68