raspberrypi/documentation #1144: External storage configuration page - umask is not supported for ext4

andyfromoz, 2019-05-15:
Step 4 of the instructions for setting up automatic mounting of an external device guide the reader to add a config line to the `fstab` file:

`UUID=5C24-1453 /mnt/mydisk FSTYPE defaults,auto,umask=000,users,rw,nofail 0 0
`

However, it seems that the `umask=000` option is not valid in fstab for `ext4` formatted devices. Perhaps a note should be added, or an alternative config line provided for ext4 formatted devices?

andyfromoz, 2019-05-15:
The "External storage configuration" page referred to is at: https://www.raspberrypi.org/documentation/configuration/external-storage.md

lurch, 2019-05-15:
Maybe this is the underlying reason behind #1051 ?

ping @andrum99 as he recently edited this page in #1122 :smiley:

andrum99, 2019-09-04:
My suggested fix for this issue would be to update that page of the docs to state that umask=000 should only be user for FAT and NTFS filesystems (and possibly Mac HFS as well). Also mention the security implications of this setting - namely that it makes all files in that filesystem fully accessible to all users.

lurch, 2019-09-06:
Fixed by #1271

closed 2019-09-06 by lurch