raspberrypi/documentation #1145: Missing documenetation for the temperature sensor hw registers

ptesarik, 2019-05-16:
The TS_TSENSCTL and TS_TSENSSTAT registers at VPU address 0x7e212000 are not documented in the BCM283x ARM Peripherals. Some things can be inferred from reading the sources of the bcm2835_thermal Linux driver contributed by Stefan Wahren, but that can hardly be called proper documentation.

ghollingworth, 2019-05-16:
Some of the registers are not documented, instead we provide support to people to write drivers (such as Stefan in this case).  The documentation would have to come from Broadcom and it has not been provided

closed 2019-05-16 by ghollingworth

ptesarik, 2019-05-16:
Shall we approach Broadcom directly. Eben Upton did not sound like the Foundation would be overly happy about it…

JamesH65, 2019-05-16:
Approaching Brcm would be entirely up to you. As would engendering the Wrath Of Eben™