raspberrypi/documentation #1148: Missing documentation of the hardware RNG

ptesarik, 2019-05-16:
The hardware random number generator (registers at VPU address 0x7e104000) is not documented. In particular, it's not clear what feeds the entropy pool in this RNG, or if it is some sort of pseudo-random generator, why it needs a warm-up and if the values read from the data registers can be considered cryptographically strong.

ghollingworth, 2019-05-16:
Cannot provide

closed 2019-05-16 by ghollingworth

ptesarik, 2019-05-16:
Hm… Since hardware RNG was not mentioned by Broadcom on a product brief for any of their embedded multimedia applications processor, for any serious applications, we should assume there is no hardware RNG on the chip.

Thanks!

JamesH65, 2019-05-16:
Assumption is wrong, there is a hardware RNG on the chip.

ghollingworth, 2019-05-16:
Yes, it uses quantum tunnelling through a band gap

ptesarik, 2019-05-16:
> Yes, it uses quantum tunnelling through a band gap

Then it shouldn't be trusted, because attackers will simply inject quantum-coupled particles into it, controlling them trough the other end of the pairs.

You made my day. :laughing:

ptesarik, 2019-05-16:
Forgot to say. Thank you for the answers. This has helped clear up some doubts.