raspberrypi/documentation #1151: Wrong documentation for Get/Set Peripheral Registers mailbox properties

ptesarik, 2019-05-17:
I'm referring to this file:

https://github.com/raspberrypi/documentation/blob/JamesH65-mailbox_docs/configuration/mailboxes/propertiesARM-VC.md

First, it seems the 0x00038045 and 0x00030045 tags are swapped (0x00030... is GET, 0x00038... is SET).

Second, it took a lot of trial and error to figure out that _the AXI bus monitor system/VPU peripherals_ actually refer to register blocks 0x7e009800–0x7e009900 and 0x7ee08000–0x7ee08100.

Third, I did not find any good documentation for these registers. Best I could find were symbolic names from a header file released by Broadcom, and they seem to be related to bus _performance_ rather than _peripherals_. Could we get a better description of these registers, please?

Fourth, the mailbox interface works differently than described, or it does not work at all. Sample output:

```
pi@raspberrypi:~$ vcgencmd version
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
pi@raspberrypi:~$ /opt/vc/bin/vcmailbox 0x00030045 12 12 0x7ee08000 1 0 0          
0x00000028 0x80000000 0x00030045 0x0000000c 0x80000003 0x7ee08000 0x00000001 0x00000001 0x00000000 0x00000000 
pi@raspberrypi:~$ /opt/vc/bin/vcmailbox 0x00030045 12 12 0x7ee08040 1 0 0 
0x00000028 0x80000000 0x00030045 0x0000000c 0x80000003 0x7ee08040 0x00000001 0x4000000d 0x00000000 0x00000000 
pi@raspberrypi:~$ /opt/vc/bin/vcmailbox 0x00030045 12 12 0x7ee08044 1 0 0 
0x00000028 0x80000000 0x00030045 0x0000000c 0x80000003 0x7ee08044 0x00000001 0x000086f4 0x00000000 0x00000000 
pi@raspberrypi:~$ /opt/vc/bin/vcmailbox 0x00030045 12 12 0x7ee08048 1 0 0 
0x00000028 0x80000000 0x00030045 0x0000000c 0x80000003 0x7ee08048 0x00000001 0x00015d64 0x00000000 0x00000000 
```

Note the bogus value length. It looks like the length is given in _words_ rather than _bytes_ (as documented). Since this is different from all other mailbox properties, it might be a bug. Should I report it against https://github.com/raspberrypi/firmware ?

JamesH65, 2019-05-17:
Note that that is pre-release documenation I knocked up one Friday afternoon, so its almost certainly got issues. When I have time I might be able to improve it, but not by much, since I don't know a lot about it. It was really intended just to get it off the Wiki in to the official docs.

lurch, 2019-05-17:
The wiki James is referring to is at https://github.com/raspberrypi/firmware/wiki
I've not checked, but it may or may not include extra information compared to James' pre-release docs branch.

ptesarik, 2019-05-17:
What about my third point? Without proper documentation of the register blocks, this API is pretty useless; it would be better to remove it both from documentation and the firmware itself.

JamesH65, 2019-05-17:
The AXI bus monitor driver uses the AXI mailbox.

ptesarik, 2019-05-17:
Fair enough. Then it would be good to fix the firmware responses. Should I report an issue against firmware?

The values reported by the AXI bus monitor may also be better explained. Looking at the first letter:
- Rtrans, Rtwait, RMax: I assume _Read_
- Wtrans, Wtwait, WMax: I assume _Write_
- Atrans, Atwait, AMax: hmm, _All_ ?

Looking at the word:
- Rtrans, Wtrans, Atrans: (Finished?) transactions
- Rtwait, Wtwait, Atwait: (Number of?) waiting transactions
- RMax, WMax, AMax: Maximum … of what?

lurch, 2019-05-18:
I did a quick google and found http://www.gstitt.ece.ufl.edu/courses/fall15/eel4720_5721/labs/refs/AXI4_specification.pdf

JamesH65, 2019-05-18:
The axi bus monitor was really only developed for internal testing purposes. I supose other might find it useful in some obscure circumstances.

ptesarik, 2019-05-18:
Excuse my perseverance, but if you found it useful enough to get a driver into the downstream kernel, then it can be probably useful to other people, too. What is the benefit of having performance counters that nobody can understand? If the counters are indeed not useful, I'd rather not have any driver for it, simply because nonexistent code cannot be buggy (unlike any other code).

lurch, 2019-05-18:
If you feel that strongly, I suppose there's nothing stopping you [compiling your own kernel](https://www.raspberrypi.org/documentation/linux/kernel/README.md) with that particular driver disabled?

ptesarik, 2019-05-19:
Oh, I should probably explain my point. I'm actually interested in running the upstream kernel on Raspberry boards (because that's the path taken by openSUSE and SUSE Linux Enterprise). The AXI bus monitor driver is available only in the foundation kernel, so it contributes to the delta between Raspbian and openSUSE for RPi. ATM I don't have any personal need for the driver, so I'm quite neutral about it, but I thought that openSUSE users might need it (or the underlying firmware interface). Given the lack of information, this does not look very likely.

To sum up the situation:
* This feature is not available in distributions that use the upstream kernel.
* The Foundation refuses to explain the meaning of values reported by this driver, hence trying to upstream it would be a waste of time.
* If anybody misses the feature in the upstream kernel, I can give them a link to this GitHub issue; they may reopen it and provide an actual use case.
* I don't care if the driver is available in the downstream kernel. Keep it if it is of any value to you.

Thanks!

closed 2019-05-19 by ptesarik

JamesH65, 2019-05-20:
Your summation is incorrect..

"The Foundation refuses to explain the meaning of values reported by this driver, hence trying to upstream it would be a waste of time."

We havent refused to do anything of the sort.  Just because we don't have time to deal with it doesn't mean we're refusing to deal with it. We are a small team with a lot of work to do.  All of which is higher priority than this..

ptesarik, 2019-05-20:
Understood. I'm sorry for drawing conclusions too early, but since the driver adds a rather thin layer on top of the hardware registers, it started to look like one of those cases where you say that Broadcom would have to provide the documentation.